google/cameraview #1: Camera2 always crashes after Permission Dialog

andhie, 2016-09-10:
1) Build and run the Demo app
2) When the Permission dialog pops up, just accept it
3) Demo app crashes immediately
--- or ---
1) Revoke the Permission from Setting and re-enter the app, 
2) after the Permission Rationale dialog is shown, just accept the Permission Dialog
3) watch it crash again.

Device: Nexus 5x running on Nougat (NRD90M)

https://github.com/google/cameraview/blob/master/library/src/main/camera2/com/google/android/cameraview/Camera2.java#L132

```java
09-11 07:05:44.562 23963-23963/com.google.android.cameraview.demo E/AndroidRuntime: FATAL EXCEPTION: main
     Process: com.google.android.cameraview.demo, PID: 23963
     java.lang.IllegalStateException: Session has been closed; further changes are illegal.
         at android.hardware.camera2.impl.CameraCaptureSessionImpl.checkNotClosed(CameraCaptureSessionImpl.java:606)
         at android.hardware.camera2.impl.CameraCaptureSessionImpl.setRepeatingRequest(CameraCaptureSessionImpl.java:227)
         at com.google.android.cameraview.Camera2$3.onConfigured(Camera2.java:132)
         at java.lang.reflect.Method.invoke(Native Method)
         at android.hardware.camera2.dispatch.InvokeDispatcher.dispatch(InvokeDispatcher.java:39)
         at android.hardware.camera2.dispatch.HandlerDispatcher$1.run(HandlerDispatcher.java:65)
         at android.os.Handler.handleCallback(Handler.java:751)
         at android.os.Handler.dispatchMessage(Handler.java:95)
         at android.os.Looper.loop(Looper.java:154)
         at android.app.ActivityThread.main(ActivityThread.java:6077)
         at java.lang.reflect.Method.invoke(Native Method)
         at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:865)
         at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:755)
```

AlirezaAkbarix, 2016-09-13:
I faced this problem too!

suryakencana, 2016-09-14:
same with me, i got this error

pzhangleo, 2016-09-20:
same error

pmbfish40, 2016-09-25:
Is anyone know how to fix it?

closed 2016-10-04 by yaraki