google/cameraview #4: Prefix library attributes

hvisser, 2016-09-11:
Because android attributes are shared in a common namespace, it's a good habit to prefix attributes like `aspectRatio` in order to prevent conflicts with other libraries that a project may use.

I'd suggest prefixing the attributes this library adds with `cam_` or `cv_` for example so that `app:aspectRatio` would become `app:cam_aspectRatio` or `app:cv_aspectRatio`.

chrisbanes, 2016-09-16:
> Because android attributes are **currently** shared in a common namespace

hvisser, 2016-09-17:
I'm not sure how far we are from `aapt2` being a thing, but I can live with that :) Filed this mainly on my experience with other libs that had clashing attributes, which is a painful experience, so I'll close this then.

closed 2016-09-17 by hvisser