google/cameraview #11: java.lang.IllegalStateException: Session has been closed; further changes are illegal.

vanniktech, 2016-09-14:
When calling `cameraView.start();` I get the following exception:

>java.lang.IllegalStateException: Session has been closed; further changes are illegal.
  at android.hardware.camera2.impl.CameraCaptureSessionImpl.checkNotClosed(CameraCaptureSessionImpl.java:606)
  at android.hardware.camera2.impl.CameraCaptureSessionImpl.setRepeatingRequest(CameraCaptureSessionImpl.java:227)
  at com.google.android.cameraview.Camera2$3.onConfigured(Camera2.java:132)
  at java.lang.reflect.Method.invoke(Native Method)
  at android.hardware.camera2.dispatch.InvokeDispatcher.dispatch(InvokeDispatcher.java:39)
  at android.hardware.camera2.dispatch.HandlerDispatcher$1.run(HandlerDispatcher.java:65)
  at android.os.Handler.handleCallback(Handler.java:751)
  at android.os.Handler.dispatchMessage(Handler.java:95)
  at android.os.Looper.loop(Looper.java:154)
  at android.app.ActivityThread.main(ActivityThread.java:6077)
  at java.lang.reflect.Method.invoke(Native Method)
  at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:865)
  at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:755)

Note: I'm not calling it in `onResume`. Rather after the Camera permisison has been granted.

vanniktech, 2016-09-14:
Seems like this is happening when the views visibility is initially set to GONE

ungesehn, 2016-09-17:
I am facing the same issue when trying the sample: on first start after granting permission the app crashes. Subsequent starts are successful and the app is working as expected. I added a longer logcat output from my device - Sony Xperia Z3 / running on Android 6.0.1

> 9-17 17:11:19.133 31706-31783/com.google.android.cameraview.demo I/OpenGLRenderer: Initialized EGL, version 1.4
> 09-17 17:11:19.258 31706-31706/com.google.android.cameraview.demo I/Timeline: Timeline: Activity_idle id: android.os.BinderProxy@8d59087 time:634371044
> 09-17 17:11:22.079 31706-31706/com.google.android.cameraview.demo I/CameraManagerGlobal: Connecting to camera service
> 09-17 17:11:22.143 31706-31712/com.google.android.cameraview.demo W/art: Suspending all threads took: 16.824ms
> 09-17 17:11:22.161 31706-31706/com.google.android.cameraview.demo I/CameraManager: Using legacy camera HAL.
> 09-17 17:11:22.669 31706-31706/com.google.android.cameraview.demo D/MainActivity: onCameraOpened
> 09-17 17:11:22.675 31706-31706/com.google.android.cameraview.demo I/CameraDeviceState: Legacy camera service transitioning to state CONFIGURING
> 09-17 17:11:22.675 31706-32161/com.google.android.cameraview.demo I/RequestThread-0: Configure outputs: 2 surfaces configured.
> 09-17 17:11:22.675 31706-32161/com.google.android.cameraview.demo D/Camera: app passed NULL surface
> 09-17 17:11:22.680 31706-32161/com.google.android.cameraview.demo I/RequestThread-0: configureOutputs - set take picture size to 5248x3936
> 09-17 17:11:22.746 31706-31706/com.google.android.cameraview.demo I/CameraDeviceState: Legacy camera service transitioning to state IDLE
> 09-17 17:11:22.757 31706-31706/com.google.android.cameraview.demo I/CameraDeviceState: Legacy camera service transitioning to state CONFIGURING
> 09-17 17:11:22.758 31706-32161/com.google.android.cameraview.demo I/RequestThread-0: Configure outputs: 2 surfaces configured.
> 09-17 17:11:22.758 31706-32161/com.google.android.cameraview.demo D/Camera: app passed NULL surface
> 09-17 17:11:22.760 31706-32161/com.google.android.cameraview.demo W/RequestThread-0: configureOutputs - Will need to crop picture 2048x1536 into smallest bound size 1600x1200
> 09-17 17:11:22.761 31706-32161/com.google.android.cameraview.demo I/RequestThread-0: configureOutputs - set take picture size to 2048x1536
> 09-17 17:11:22.780 31706-31706/com.google.android.cameraview.demo I/CameraDeviceState: Legacy camera service transitioning to state IDLE
> 09-17 17:11:22.783 31706-31706/com.google.android.cameraview.demo W/PathParser: Points are too far apart 4.000000596046461
> 09-17 17:11:22.786 31706-31706/com.google.android.cameraview.demo D/AndroidRuntime: Shutting down VM
> 09-17 17:11:22.787 31706-31706/com.google.android.cameraview.demo E/AndroidRuntime: FATAL EXCEPTION: main
>                                                                                     Process: com.google.android.cameraview.demo, PID: 31706
>                                                                                     java.lang.IllegalStateException: Session has been closed; further changes are illegal.
>                                                                                         at android.hardware.camera2.impl.CameraCaptureSessionImpl.checkNotClosed(CameraCaptureSessionImpl.java:606)
>                                                                                         at android.hardware.camera2.impl.CameraCaptureSessionImpl.setRepeatingRequest(CameraCaptureSessionImpl.java:227)
>                                                                                         at com.google.android.cameraview.Camera2$3.onConfigured(Camera2.java:132)
>                                                                                         at java.lang.reflect.Method.invoke(Native Method)
>                                                                                         at android.hardware.camera2.dispatch.InvokeDispatcher.dispatch(InvokeDispatcher.java:39)
>                                                                                         at android.hardware.camera2.dispatch.HandlerDispatcher$1.run(HandlerDispatcher.java:65)
>                                                                                         at android.os.Handler.handleCallback(Handler.java:739)
>                                                                                         at android.os.Handler.dispatchMessage(Handler.java:95)
>                                                                                         at android.os.Looper.loop(Looper.java:234)
>                                                                                         at android.app.ActivityThread.main(ActivityThread.java:5526)
>                                                                                         at java.lang.reflect.Method.invoke(Native Method)
>                                                                                         at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:726)
>                                                                                         at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:616)

stoyicker, 2016-09-17:
Note that this also seems to happen when you put your .stop call AFTER the super call in your onPause, instead of before. Could be the issue here as well.

gilshallem, 2016-09-29:
Same problem happen to me on nexus 5x 7.0

gilshallem, 2016-09-29:
Here is a temp workaround:
in camera2.java -> add this in line 134
` } catch (IllegalStateException e) {
                if (e.getMessage().contains("Session has been closed")) {
                    start();
                }`

Ghedeon, 2016-10-04:
Happens to me if you lock/unlock the screen. Same code added to a different empty Activity works just fine. Yet didn't find what so special about the first one.

arriolac, 2017-02-22:
Seeing this as well.

arriolac, 2017-02-22:
When I remove the callback in mPreview I no longer get this exception (i.e. remove https://github.com/google/cameraview/blob/master/library/src/main/api21/com/google/android/cameraview/Camera2.java#L191...L196). Is this safe to remove or would that have unintended consequences?

TarekAshrafAhmed, 2017-03-20:
try to update to compile "com.android.support:support-v4:25.2.0"
finally it worked with me

mattfred, 2017-05-10:
Is this still being looked into? I'm getting this same error.

oyra, 2017-06-30:
I can see the pull request, but is it going to be merged? The bug still exists

instantiator, 2017-08-18:
It seems like this was occurring for me when I `setVisibility(VISIBLE)` and then immediately attempted to `start()` the `CameraView`. My workaround: do the visibility change and `post` the `start()` to allow it time for the view to be ready. Seems to work!

    view_main_camera.setVisibility(VISIBLE);
    view_main_camera.post(new Runnable() {
      @Override
      public void run() {
        view_main_camera.start();
      }
    });

jmarkstar, 2017-09-28:
Well, after try many ways, I got a kind of solution.

I just wanted when I stop the camera, I also want to hide the CameraView.

My first trying was like that, and I got the issue **Session has been closed; further changes are illegal**.

    mCameraView.setVisibility(View.VISIBLE);
    mCameraView.start();

    mCameraView.stop();
    mCameraView.setVisibility(View.GONE);

Then, I just had a look at the callback **CameraView.Callback** and I realised that it has two methods onCameraOpened() and onCameraClosed(). I just put the setVisibles in those methods and I didn't get more issues anymore.

    private CameraView.Callback mCallback = new CameraView.Callback() {
    
            @Override public void onCameraOpened(CameraView cameraView) {
                Timber.v("onCameraOpened");
                mCameraView.setVisibility(View.VISIBLE);
            }
    
            @Override public void onCameraClosed(CameraView cameraView) {
                Timber.v("onCameraClosed");
                mCameraView.setVisibility(View.GONE);
            }
    
            @Override public void onPictureTaken(CameraView cameraView, final byte[] data) {}
    }

Sorry for my english.

ivanTrogrlic, 2017-10-11:
I guess some devices don't support some aspect ratios. The default aspect ratio is 4:3 and I was getting this error on LG G4. I tried changing the aspect ratio in `onCameraOpened` callback to 16:9 and it worked fine. 

The code:

`        @Override
        public void onCameraOpened(CameraView cameraView) {
            mCameraView.setAspectRatio(AspectRatio.of(16, 9));
        }`

vanniktech, 2017-12-15:
Ended up switching to [Fotoapparat](https://github.com/Fotoapparat/Fotoapparat).

closed 2017-12-15 by vanniktech

zweinz, 2018-01-08:
Was this closed just because the reporter ended up using a different solution?  I'm still seeing this in my own project running on a Pixel.

einschneidend, 2018-06-12:
No success here, what I had to do (no time as always) is that I implemented a onCrash() callback
so if it happens, I restart the camera activity and now all works well.

Yes yes, I know.