google/cameraview #17: Bug on Note 4

EMIN3M1986, 2016-09-18:
Hi.
I'm using you library.
In some cellphones like  Samsung Note 4 when I use front camera it dose not work with this Log: 

15726-23967/com.google.android.cameraview.demo W/LegacyRequestMapper: convertRequestMetadata - control.awbRegions setting is not supported, ignoring value
15726-23967/com.google.android.cameraview.demo W/LegacyMetadataMapper: convertAfModeToLegacy - ignoring unsupported mode 4, defaulting to fixed
15726-23967/com.google.android.cameraview.demo W/LegacyRequestMapper: convertRequestMetadata - control.awbRegions setting is not supported, ignoring value
15726-23967/com.google.android.cameraview.demo W/LegacyMetadataMapper: convertAfModeToLegacy - ignoring unsupported mode 4, defaulting to fixed

linhnguyen106, 2016-10-19:
I got the same problem. is this going to be fixed ?

jaksab, 2016-11-11:
+1 Help please

fabcam, 2017-01-19:
did any of you solved this issue?? thanks!