google/cameraview #20: Aspect Ratio not taken into account

bunjix, 2016-09-19:
I have tried with aspectRatio="1:1" but nothing change.

yaraki, 2016-09-21:
Because it is not one of the supported aspect ratios. We will figure out how to deliver errors properly. For now, please make sure to use CameraView#getSupportedAspectRatios before setting an aspect ratio.

ivacf, 2016-09-22:
@yaraki so if we always want our camera view to be a square but the camera doesn't support a 1:1 ratio, what's your recommended way to achieve this?

yaraki, 2016-09-26:
I'll investigate.

remq, 2016-10-05:
Couldn't you just put the CameraView inside a square parent? Something like this:

```java
public class SquareRelativeLayout extends RelativeLayout {

    public SquareRelativeLayout(Context context, AttributeSet attrs) {
        super(context, attrs);
    }

    @Override
    protected void onMeasure(int widthMeasureSpec, int heightMeasureSpec) {
        super.onMeasure(heightMeasureSpec , heightMeasureSpec );
    }
}
```

yaraki, 2016-10-05:
That will crop the CameraView to square size, but the taken pictures will not be square.

This is a combination of 2 issues. I'm working on a fix.

- Aspect Ratio is not working properly.
- Nexus 5X reports 1:1 ratio as one of the supported picture format, but the preview crashes with SurfaceView.

bunjix, 2016-10-16:
@yaraki Do you have any update on this ?

yaraki, 2016-10-18:
Not yet. I'm working on it.

yaraki, 2016-12-15:
I added aspect ratio choice in the demo app. Also, setAspectRatio now works after layout. I still need to fix the issue with 1:1 on N5x and such.

closed 2017-04-21 by yaraki

bolein, 2017-06-22:
Nexus 6P, Android Nougat, 1:1 aspect ratio is ignored

iftekhar-ahmed, 2017-07-26:
I get the same issue with nexus 5x android nougat/O. can't get a 1:1 setting it with a setAspectRatio call after layout. Is there an update to this problem?

drottemberg, 2017-12-06:
I had the same issue and basically you need to set the aspect ratio after the camera has been initialized in this function

```private CameraView.Callback mCallback
            = new CameraView.Callback() {

        @Override
        public void onCameraOpened(CameraView cameraView) {
            Set s = mCameraView.getSupportedAspectRatios(); // This is just to make sure the set is not empty
            mCameraView.setAspectRatio(AspectRatio.of(1,1));
        }
```

jgarcia3327, 2017-12-15:
Do you guys have a fix on the issue? Have white background appearing on my Nexus 5x running Android 8 on 16:9 preview. All other phone I tested running ok. Thanks.

nawinkhatiwada, 2018-10-24:
could not achieve aspect ratio 1:1 on Nokia 6

Himanshu507, 2020-04-10:
> Do you guys have a fix on the issue? Have white background appearing on my Nexus 5x running Android 8 on 16:9 preview. All other phone I tested running ok. Thanks.

Hey can you help me. i'm recording the video on full screen but when i preview this video it not plays in full screen why ?

commit 9bbe07ba (2017-04-21): Fix aspect ratio of taken pictures

Fixes #127
Fixes #78
Fixes #46
Fixes #20

Change-Id: I93f62823f3c1fd8a28ee4b659a9a5b7c81f183fc