google/cameraview #22: Wrong picture orientation on Samsung devices

ivacf, 2016-09-26:
Hi,

After taking a picture with this library I create a Bitmap and set it in an ImageView as follow:

```java
 public void onPictureTaken(CameraView cameraView, byte[] data) {
     Bitmap bitmap = BitmapFactory.decodeByteArray(data, 0, data.length);
     imageView.setImageBitmap(bitmap);
}
```

This works fine on a Nexus 5X, however on Samsung devices the picture orientation appears to be incorrectly rotated 90 degrees to the left when the device orientation is portrait. Is there a way to fix this issue?

rockar06, 2016-10-05:
LG-K500h (X screen)  has the same behaviour, after the photo is taken, is rotated 90 degrees to the left.

xCatG, 2016-10-27:
This is because Samsung (also Sony) devices write EXIF orientation = 6 but `BitmapFactory.decodeByteArray()` ignores the EXIF rotation tag. (I think it will work if you write the `data` out to a file and then read it using `BitmapFactory.decodeFile`) You have to rotate the image yourself before displaying it.

samy-baili, 2016-11-28:
Did someone find a workaround ?

damianMolinski, 2016-11-28:
I've created pull request[https://github.com/google/cameraview/pull/45/files] which adds option to have, as result, byte array and angle of image so u can easily rotate bitmap.

All u have to do is copy `ExifUtil` class and then use `getOrientation` with byte array. 
Then u can rotate bitmap, e.g.
`private Bitmap rotateBitmap(Bitmap source, int angle) {
         Matrix matrix = new Matrix();
         matrix.postRotate(angle);
         return Bitmap.createBitmap(source, 0, 0, source.getWidth(), source.getHeight(), matrix, true);
 }`
 
I've tested it on samsung galaxy s6/nexus5 and it does work for me.

linhnguyen106, 2016-12-26:
is anyone gonna fix this ?

ivacf, 2016-12-27:
I don't think this is a issue with the library anymore, the application should be in charge of rotating the image if necessary by reading the image Exif data. This can be easily achieved using the new [Exif support library](https://android-developers.googleblog.com/2016/12/introducing-the-exifinterface-support-library.html)

The exif library allows you to easily find out how much you have to rotate the image before you display it to the user. This is how it could be done:

```java 
@Override
public void onPictureTaken(CameraView cameraView, byte[] data) {
            // Find out if the picture needs rotating by looking at its Exif data
            ExifInterface exifInterface = new ExifInterface(new ByteArrayInputStream(data));
            int orientation = exifInterface.getAttributeInt(ExifInterface.TAG_ORIENTATION, 1);
            int rotationDegrees = 0;
            switch (orientation) {
                case ExifInterface.ORIENTATION_ROTATE_90:
                    rotationDegrees = 90;
                    break;
                case ExifInterface.ORIENTATION_ROTATE_180:
                    rotationDegrees = 180;
                    break;
                case ExifInterface.ORIENTATION_ROTATE_270:
                    rotationDegrees = 270;
                    break;
            }
            // Create and rotate the bitmap by rotationDegrees
}
```

I think this issue can be closed now.

petergayduchik, 2017-11-14:
its still not working . more than 5 years. And bug still here. I cant believe it . Samsungs devices have a lot of bugs. I cant understand why they cant fix this problem.

sibelius, 2017-12-10:
you should follow @ivacf comment https://github.com/google/cameraview/issues/22#issuecomment-269321811

mohitajwani, 2018-02-05:
@ivacf I am using this code. `normalizeImageForUri` method is used to fix these things. But, I am getting a problem in Nougat device. I am not sure if the same issue persists in lower versions. The issue is I have sent the `Uri` to this method by calling `Uri.fromFile(photoFile)` where `photoFile` is the file that is made from the Util class and Uri is extracted using this code `FileProvider.getUriForFile(mCallerActivity,
                                BuildConfig.APPLICATION_ID + ".provider",
                                mPhotoFile);`

I am unable get the ExifInterface object, I get a FileNotFoundException. Whereas, if I am using the same file to upload the image to my S3 bucket and that is successful. Can you please let me know why the ExifInterface object is not getting the File?

`
public static void normalizeImageForUri(Context context, Uri uri) {
        try {
            ExifInterface exif = new ExifInterface(uri.getPath());
            int orientation = exif.getAttributeInt(ExifInterface.TAG_ORIENTATION, ExifInterface.ORIENTATION_UNDEFINED);
            Bitmap bitmap = MediaStore.Images.Media.getBitmap(context.getContentResolver(), uri);
            Bitmap rotatedBitmap = rotateBitmap(bitmap, orientation);
            if (!bitmap.equals(rotatedBitmap)) {
                saveBitmapToFile(context, rotatedBitmap, uri);
            }
        } catch (IOException e) {
            e.printStackTrace();
        }
    }

    private static Bitmap rotateBitmap(Bitmap bitmap, int orientation) {
        Matrix matrix = new Matrix();
        switch (orientation) {
            case ExifInterface.ORIENTATION_NORMAL:
                return bitmap;
            case ExifInterface.ORIENTATION_FLIP_HORIZONTAL:
                matrix.setScale(-1, 1);
                break;
            case ExifInterface.ORIENTATION_ROTATE_180:
                matrix.setRotate(180);
                break;
            case ExifInterface.ORIENTATION_FLIP_VERTICAL:
                matrix.setRotate(180);
                matrix.postScale(-1, 1);
                break;
            case ExifInterface.ORIENTATION_TRANSPOSE:
                matrix.setRotate(90);
                matrix.postScale(-1, 1);
                break;
            case ExifInterface.ORIENTATION_ROTATE_90:
                matrix.setRotate(90);
                break;
            case ExifInterface.ORIENTATION_TRANSVERSE:
                matrix.setRotate(-90);
                matrix.postScale(-1, 1);
                break;
            case ExifInterface.ORIENTATION_ROTATE_270:
                matrix.setRotate(-90);
                break;
            default:
                return bitmap;
        }
        try {
            Bitmap bmRotated = Bitmap.createBitmap(bitmap, 0, 0, bitmap.getWidth(), bitmap.getHeight(), matrix, true);
            bitmap.recycle();

            return bmRotated;
        } catch (OutOfMemoryError e) {
            e.printStackTrace();
            return null;
        }
    }

    private static void saveBitmapToFile(Context context, Bitmap croppedImage, Uri saveUri) {
        if (saveUri != null) {
            OutputStream outputStream = null;
            try {
                outputStream = context.getContentResolver().openOutputStream(saveUri);
                if (outputStream != null) {
                    croppedImage.compress(Bitmap.CompressFormat.JPEG, 90, outputStream);
                }
            } catch (IOException e) {

            } finally {
                closeSilently(outputStream);
                croppedImage.recycle();
            }
        }
    }

    private static void closeSilently(@Nullable Closeable c) {
        if (c == null) {
            return;
        }
        try {
            c.close();
        } catch (Throwable t) {
            // Do nothing
        }
    }
`

sibelius, 2018-02-05:
you can check https://github.com/react-native-community/react-native-camera

tomtheunit, 2018-02-16:
@ivacf does your code require API level 24?

MrBrown09, 2018-07-26:
@MoustafaElsaghier - Thanks for this solution. I have implemented this and tested. Working good on Samsung J pro but image turns upside down(i.e. 180 degree) in Samsung J Core. Don't know why it is. Please suggest a solution.

MoustafaElsaghier, 2018-07-26:
@MrBrown09 sorry for the late answer
I've solved this by using **```OrientationEventListener```** and in **```onOrientationChanged```** I reversed the orientation that means if it is 0 put it 180 and if 180 set it 0

the same with 270 and 90. 

if you need any help how calc angel tell me, please

weichao66666, 2018-08-03:
关键原因：bitmap 不包含 exif 信息
解决方法1：读文件前获取 exif 信息，bitmap 写入到文件以后写入 exif 信息。
    public static ExifInterface getExif(String filePath) {
        try {
            return new ExifInterface(filePath);
        } catch (IOException e) {
            e.printStackTrace();
        }
        return null;
    }

    public static void setExif(String filePath, ExifInterface exif) {
        try {
            ExifInterface newExif = new ExifInterface(filePath);
            final Class<ExifInterface> cls = ExifInterface.class;
            final Field[] fields = cls.getFields();
            for (Field field : fields) {
                final String fieldName = field.getName();
                if (!TextUtils.isEmpty(fieldName) && fieldName.startsWith("TAG")) {
                    final String tag = field.get(cls).toString();
                    final String value = exif.getAttribute(tag);
                    if (value != null) {
                        newExif.setAttribute(tag, value);
                    }
                }
            }
            newExif.saveAttributes();
        } catch (Exception e) {
            e.printStackTrace();
        }
    }
解决方法2：读取 exif 信息，按 exif 信息旋转，最后一定不能再写入原来的ExifInterface.TAG_ORIENTATION的值，该值旋转后应该为0。
    public static Bitmap resetOrientation(String filePath) {
        if (TextUtils.isEmpty(filePath)) {
            return null;
        }

        final Bitmap bitmap = decodeBitmapFromFile(filePath);// 略
        final int degree = getOrientationDegree(filePath);
        return rotateBitmap(bitmap, degree);
    }

    public static int getOrientationDegree(String imagePath) {
        if (TextUtils.isEmpty(imagePath)) {
            return 0;
        }

        try {
            final ExifInterface exifInterface = new ExifInterface(imagePath);
            final int attributeInt = exifInterface.getAttributeInt(ExifInterface.TAG_ORIENTATION, ExifInterface.ORIENTATION_UNDEFINED);
            switch (attributeInt) {
                case ExifInterface.ORIENTATION_ROTATE_90:
                    return 90;
                case ExifInterface.ORIENTATION_ROTATE_180:
                    return 180;
                case ExifInterface.ORIENTATION_ROTATE_270:
                    return 270;
            }
        } catch (IOException e) {
            e.printStackTrace();
        }
        return 0;
    }

    public static Bitmap rotateBitmap(Bitmap bitmap, int degree) {
        if (degree == 0 || bitmap == null) {
            return bitmap;
        }

        final Matrix matrix = new Matrix();
        matrix.setRotate(degree, bitmap.getWidth() / 2, bitmap.getHeight() / 2);
        return Bitmap.createBitmap(bitmap, 0, 0, bitmap.getWidth(), bitmap.getHeight(), matrix, true);
    }

petergayduchik, 2018-09-03:
anyone fixed this bullshit for samsung android 8 + ?? (for example on samsung s8 with OS 8.0 + its  stilll returned bad rotation)

Rohit-Me-Redmart, 2018-09-03:
@petergayduchik able to solve the issue by this.

you have to rotate image by 90 degree.

`[@Override
public void onPictureTaken(CameraView cameraView, byte[] data) {
// Find out if the picture needs rotating by looking at its Exif data
ExifInterface exifInterface = new ExifInterface(new ByteArrayInputStream(data));
int orientation = exifInterface.getAttributeInt(ExifInterface.TAG_ORIENTATION, 1);
int rotationDegrees = 0;
switch (orientation) {
case ExifInterface.ORIENTATION_ROTATE_90:
rotationDegrees = 90;
break;
case ExifInterface.ORIENTATION_ROTATE_180:
rotationDegrees = 180;
break;
case ExifInterface.ORIENTATION_ROTATE_270:
rotationDegrees = 270;
break;

          default:
                matrix.postRotate(IMAGE_ROTATION_ANGLE_90);
                break;
        }
        // Create and rotate the bitmap by rotationDegrees
}](url)`

petergayduchik, 2019-05-02:
i hate developers on Samsung side . It's perfect idiotism . 
How the f...k it's possible to create this stupid system : 
Photo taked in portrate /landscape mode always have logic where WIDTH > HEIGHT(for example 4000x3000)
every time
**From Camera :**
Portate camera mode : exif = 0, photo rotated on 90 degree in left way  (incorrect)
Landscape camera mode : exif =0, photo in normal view  (correct)

**From Gallery :**
Portrate photo : exif = 90 , photo rotated on 90 degree in left way (incorrect)
Landscape photo: exif = 90 , photo in normal view (correct)

Bonten156, 2019-06-20:
I have the same problem bug it is for face detection with camera2 API, my app is blocked in portrait mode and I don't now why, with a samsung Galaxy note 8, face detection only work in landscape mode... Do someone know how to fix it ?

WaheedNazir, 2019-10-01:
@Bonten156 I came across this blog, might be helpful for you.

https://medium.com/@kenodoggy/solving-image-rotation-on-android-using-camera2-api-7b3ed3518ab6

amkrys, 2021-03-08:
Anyone fixed this in Android 11 Samsung Device?

Android563, 2021-03-08:
@amkrys is [this](https://github.com/google/cameraview/issues/22#issuecomment-383416821) work for you?

amkrys, 2021-04-01:
> @amkrys is [this](https://github.com/google/cameraview/issues/22#issuecomment-383416821) work for you?

no but now am using https://github.com/Dhaval2404/ImagePicker this library and its working fine for Samsung devices too