google/cameraview #24: Demo using Fragment

skywaydave, 2016-09-28:
Hi There,

I'm embedding the cameraview in a Fragment (within an Activity) and have run into the following exception when mCameraView.start() is called in onResume():

    java.lang.IllegalStateException: Session has been closed; further changes are illegal.
    at android.hardware.camera2.impl.CameraCaptureSessionImpl.checkNotClosed(CameraCaptureSessionImpl.java:606)
    at android.hardware.camera2.impl.CameraCaptureSessionImpl.setRepeatingRequest(CameraCaptureSessionImpl.java:227)
    at com.google.android.cameraview.Camera2$3.onConfigured(Camera2.java:132)
    at java.lang.reflect.Method.invoke(Native Method)
    at android.hardware.camera2.dispatch.InvokeDispatcher.dispatch(InvokeDispatcher.java:39)
    at android.hardware.camera2.dispatch.HandlerDispatcher$1.run(HandlerDispatcher.java:65)
    at android.os.Handler.handleCallback(Handler.java:739)
    at android.os.Handler.dispatchMessage(Handler.java:95)
    at android.os.Looper.loop(Looper.java:148)
    at android.app.ActivityThread.main(ActivityThread.java:5417)
    at java.lang.reflect.Method.invoke(Native Method)
    at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:726)
    at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:616)

I'm running on a Nexus 6P with Android 6.0.

Is there a working example that illustrates how to do this?

Thanks

yaraki, 2016-09-29:
I haven't tried this myself yet. We should definitely enhance the demo app with more use cases like this.

stoyicker, 2016-09-29:
This is how I made it work in a Fragment:

 ```java
   @Override
    public void onResume() {
        super.onResume();

        if (cameraView != null && isCapturing) {
            cameraView.start();
        }
    }

    @Override
    public void onPause() {
        if (cameraView != null && isCapturing)
            cameraView.stop();

        super.onPause();
    }
```

The flag is manually set when calling start/stop.