google/cameraview #27: Flaky capturing

Ghedeon, 2016-10-01:
Not every single click results in onPictureTaken() callback.

I'm not talking about throttling when you click too fast, as you can see, it's enough time between clicks. It's just skipped randomly.

OnePlus One, 6.0.1

` 18:30:24.111  D/MainActivity: onClick: 
 18:30:24.158  W/LegacyRequestMapper: convertRequestMetadata - control.awbRegions setting is not supported, ignoring value
 18:30:24.158  W/LegacyRequestMapper: Only received metering rectangles with weight 0.
 18:30:24.158  W/LegacyRequestMapper: Only received metering rectangles with weight 0.
 18:30:24.211  D/LegacyFocusStateMapper: onAutoFocusMoving - ignoring move callbacks from old af run1
 18:30:24.213  W/LegacyRequestMapper: convertRequestMetadata - control.awbRegions setting is not supported, ignoring value
 18:30:24.213  W/LegacyRequestMapper: Only received metering rectangles with weight 0.
 18:30:24.213  W/LegacyRequestMapper: Only received metering rectangles with weight 0.
 18:30:24.269  I/RequestQueue: Repeating capture request cancelled.
 18:30:24.302  W/LegacyRequestMapper: convertRequestMetadata - control.awbRegions setting is not supported, ignoring value
 18:30:24.303  W/LegacyRequestMapper: Only received metering rectangles with weight 0.
 18:30:24.303  W/LegacyRequestMapper: Only received metering rectangles with weight 0.
 18:30:24.883  I/RequestThread-0: Received jpeg.
 18:30:24.884  I/RequestThread-0: Producing jpeg buffer...
 18:30:24.887  D/ImageReader_JNI: ImageReader_lockedImageSetup: Receiving JPEG in HAL_PIXEL_FORMAT_RGBA_8888 buffer.
 18:30:24.889  D/MainActivity: onPictureTaken 1026780
 18:30:24.922  I/RequestQueue: Repeating capture request set.
 18:30:24.924  W/LegacyRequestMapper: convertRequestMetadata - control.awbRegions setting is not supported, ignoring value
 18:30:24.924  W/LegacyRequestMapper: Only received metering rectangles with weight 0.
 18:30:24.924  W/LegacyRequestMapper: Only received metering rectangles with weight 0.
 18:30:25.053  D/LegacyFocusStateMapper: onAutoFocusMoving - ignoring move callbacks from old af run2
 18:30:25.155  W/LegacyRequestMapper: convertRequestMetadata - control.awbRegions setting is not supported, ignoring value
 18:30:25.155  W/LegacyRequestMapper: Only received metering rectangles with weight 0.
 18:30:25.157  W/LegacyRequestMapper: Only received metering rectangles with weight 0.

 18:30:29.720  D/MainActivity: onClick: 
 18:30:29.771  W/LegacyRequestMapper: convertRequestMetadata - control.awbRegions setting is not supported, ignoring value
 18:30:29.772  W/LegacyRequestMapper: Only received metering rectangles with weight 0.
 18:30:29.772  W/LegacyRequestMapper: Only received metering rectangles with weight 0.
 18:30:29.796  W/LegacyRequestMapper: convertRequestMetadata - control.awbRegions setting is not supported, ignoring value
 18:30:29.796  W/LegacyRequestMapper: Only received metering rectangles with weight 0.
 18:30:29.796  W/LegacyRequestMapper: Only received metering rectangles with weight 0.

 18:30:36.730  D/MainActivity: onClick: 
 18:30:36.796  W/LegacyRequestMapper: convertRequestMetadata - control.awbRegions setting is not supported, ignoring value
 18:30:36.796  W/LegacyRequestMapper: Only received metering rectangles with weight 0.
 18:30:36.796  W/LegacyRequestMapper: Only received metering rectangles with weight 0.
 18:30:36.818  D/LegacyFocusStateMapper: onAutoFocusMoving - ignoring move callbacks from old af run4
 18:30:36.872  W/LegacyRequestMapper: convertRequestMetadata - control.awbRegions setting is not supported, ignoring value
 18:30:36.872  W/LegacyRequestMapper: Only received metering rectangles with weight 0.
 18:30:36.872  W/LegacyRequestMapper: Only received metering rectangles with weight 0.
 18:30:36.873  I/RequestQueue: Repeating capture request cancelled.
 18:30:36.941  W/LegacyRequestMapper: convertRequestMetadata - control.awbRegions setting is not supported, ignoring value
 18:30:36.941  W/LegacyRequestMapper: Only received metering rectangles with weight 0.
 18:30:36.941  W/LegacyRequestMapper: Only received metering rectangles with weight 0.
 18:30:37.702  I/RequestThread-0: Received jpeg.
 18:30:37.702  I/RequestThread-0: Producing jpeg buffer...
 18:30:37.703  D/ImageReader_JNI: ImageReader_lockedImageSetup: Receiving JPEG in HAL_PIXEL_FORMAT_RGBA_8888 buffer.
 18:30:37.704  D/MainActivity: onPictureTaken 1022546

`

Ghedeon, 2016-10-04:
Looks like same issue is already reported for the official Camera2Basic sample app: https://github.com/googlesamples/android-Camera2Basic/issues/33

aseem-khare, 2018-01-03:
The issue is with your android device. The devices that do not support auto focus throw this error. You need to disable auto focus to get rid of this error.

delacrixmorgan, 2019-03-11:
There's a few reasons.

a) While you are `capturePicture()`, the AutoFocus state is stucked at `CONTROL_AF_STATE_PASSIVE_FOCUSED` and there is no way to shake it off.

Thus, you'll need to manually restart the focus and get a lock on it again.

b) `setRepeatingRequest()` from `captureSession` is overwhelming it. 

To resolve that, you'll need to add a delay.

Step 1 - Restructure `capturePicture(result: CaptureResult)`

```kotlin
private fun capturePicture(result: CaptureResult) {
    val afState = result.get(CaptureResult.CONTROL_AF_STATE)
    when (afState) {
        CaptureResult.CONTROL_AF_STATE_FOCUSED_LOCKED, CaptureResult.CONTROL_AF_STATE_NOT_FOCUSED_LOCKED -> {
            val aeState = result.get(CaptureResult.CONTROL_AE_STATE)
            if (aeState == null || aeState == CaptureResult.CONTROL_AE_STATE_CONVERGED) {
                state = STATE_PICTURE_TAKEN
                captureStillPicture()
            } else {
                runPrecaptureSequence()
            }
        }
        null -> captureStillPicture()
        else -> restartFocus()
    }
}
```

Step 2 - Add `restartFocus()`

```kotlin
private fun restartFocus() {
    try {
        previewRequestBuilder.set(CaptureRequest.CONTROL_AF_TRIGGER, CameraMetadata.CONTROL_AF_TRIGGER_CANCEL)
        captureSession?.capture(previewRequestBuilder.build(), captureCallback, backgroundHandler)
        lockFocus()
    } catch (e: CameraAccessException) {
        Log.e(CameraFragment::class.java.simpleName, e.toString())
    }
}
```

Step 3 - Add Delay for `repeatingRequest` inside `unlockFocus()`

```kotlin
private fun unlockFocus() {
    try {
        // Reset the auto-focus trigger
        previewRequestBuilder.set(CaptureRequest.CONTROL_AF_TRIGGER, CameraMetadata.CONTROL_AF_TRIGGER_CANCEL)
        captureSession?.capture(previewRequestBuilder.build(), captureCallback, backgroundHandler)
        // After this, the camera will go back to the normal state of preview.
        state = STATE_PREVIEW
        Handler().postDelayed({
            captureSession?.setRepeatingRequest(previewRequest, captureCallback, backgroundHandler)
        }, 500)
    } catch (e: CameraAccessException) {
        Log.e(CameraFragment::class.java.simpleName, e.toString())
    }
}
```
Special thanks to these StackOverFlow discussion that enabled me to figure it out.

https://stackoverflow.com/questions/44527420/android-camera2-focus-state-stuck

https://stackoverflow.com/questions/40146485/session-has-been-closed-further-changes-are-illegal