google/cameraview #29: NULL reference exception on takePicture()

remq, 2016-10-04:
I only got this error on my Huawei Ascend P7 device and not on a Samsung Galaxy S5. When taking a picture, it crashes.

`mTakePicture.setOnClickListener(new View.OnClickListener() {
    @Override
    public void onClick(View v) {
        if (mCameraView.isCameraOpened()) {
            mCameraView.takePicture();
        }
    }
});

mCameraView.addCallback(new CameraView.Callback() {
    @Override
    public void onPictureTaken(CameraView cameraView, byte[] data) {
        super.onPictureTaken(cameraView, data);
        mListener.onPictureTaken(data); // Custom method to process the data
        mCameraView.stop();
        getActivity().getSupportFragmentManager().popBackStack();
    }
});
`

When deleting the following line it works, however I need to close the camera fragment.
`getActivity().getSupportFragmentManager().popBackStack();`

This is the error:

`java.lang.NullPointerException: Attempt to invoke virtual method 'int android.hardware.camera2.CameraCaptureSession.capture(android.hardware.camera2.CaptureRequest, android.hardware.camera2.CameraCaptureSession$CaptureCallback, android.os.Handler)' on a null object reference
    at com.google.android.cameraview.Camera2.unlockFocus(Camera2.java:625)
    at com.google.android.cameraview.Camera2.access$900(Camera2.java:45)
    at com.google.android.cameraview.Camera2$6.onCaptureCompleted(Camera2.java:609)
    at java.lang.reflect.Method.invoke(Native Method)
    at java.lang.reflect.Method.invoke(Method.java:372)
    at android.hardware.camera2.dispatch.InvokeDispatcher.dispatch(InvokeDispatcher.java:39)
    at android.hardware.camera2.dispatch.HandlerDispatcher$1.run(HandlerDispatcher.java:65)
    at android.os.Handler.handleCallback(Handler.java:739)
    at android.os.Handler.dispatchMessage(Handler.java:95)
    at android.os.Looper.loop(Looper.java:135)
    at android.app.ActivityThread.main(ActivityThread.java:5569)
    at java.lang.reflect.Method.invoke(Native Method)
    at java.lang.reflect.Method.invoke(Method.java:372)
    at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:931)
    at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:726)`

My guess is that I need to listen to another callback before killing the camera, but I couldn't find one.

remq, 2016-10-05:
I've fixed it for myself by adding a confirm-picture screen. This way the camera has enough time to close itself and it's a good practice anyway. However, this issue might be worth looking into.

PavelSynek, 2016-11-01:
I solved the problem by calling `camera.stop()` instead of navigating away and overriding `onCameraClosed` in `CameraView.Callback`. I wait until the camera is closed and then perform the navigation.

PeterJiung, 2016-11-15:
i have the same bug

jinqian, 2017-01-18:
FYI, I have the same bug on Samsung Galaxy S7 and S5.

JMoicano, 2018-08-23:
https://github.com/PavelSynek can you give a little example for this solution?

JMoicano, 2018-08-23:
@PavelSynek can you give a sample of how your sollution works?

PavelSynek, 2018-08-23:
Sure! I don't think I can find the exact code, but this pseudocode should help too.

```java
private boolean navigateAfterCameraStop = false;

cameraView.addCallback(new CameraView.Callback() {
    @Override
    public void onPictureTaken(CameraView cameraView, byte[] data) {
         //
    }

    @Override
    public void onCameraClosed() {
        if (navigateAfterCameraStop) {
            navigateAfterCameraStop = false;
            navigateForReal();
        }
    }
});

public void onNavigate() {
    // Do not navigate, but set the flag, stop camera, wait for it to close and then navigate
    navigateAfterCameraStop = true;
    cameraView.stop();
}

```

JMoicano, 2018-08-28:
@PavelSynek what is the event calling onNavigate? It's called in navigateForReal? Sorry for this dumb questions, I'm stumbling on this matter

M0N3, 2018-10-11:
Just added 
catch (NullPointerException n) {
            n.printStackTrace();
        }
to method unlockFocus()
in Camera2.java class
Dirty trick, but works.

siokagami, 2019-06-24:
> Just added
> 
> ```
> catch (NullPointerException n) {
>             n.printStackTrace();
>         }
> ```
> 
> to method unlockFocus()
> in Camera2.java class
> Dirty trick, but works.

 Really work for me！！！！

yink-liu, 2019-09-07:
> Just added
> 
> ```
> catch (NullPointerException n) {
>             n.printStackTrace();
>         }
> ```
> 
> to method unlockFocus()
> in Camera2.java class
> Dirty trick, but works.

> > Just added
> > ```
> > catch (NullPointerException n) {
> >             n.printStackTrace();
> >         }
> > ```
> > 
> > 
> > to method unlockFocus()
> > in Camera2.java class
> > Dirty trick, but works.
> 
> Really work for me！！！！

+1

andzejsw, 2020-02-24:
Same error on Xiaomi: Redmi Note 7.

pankaj529, 2020-08-20:
@andzejsw  got any solution??