google/cameraview #30: Bug on Sony Xperia Z2

WeRockStar, 2016-10-06:
After take a picture incorrectly rotated bitmap.

yaraki, 2016-10-14:
The provided information is not enough. I don't have that device, so there is currently no way I can fix this.

ivacf, 2016-10-15:
I think this issue is related to https://github.com/google/cameraview/issues/22, Several non-Nexus devices seem to have issues with image orientation. I had a look at the code that handles image orientation and I couldn't see anything wrong with it. I also noticed that these orientation problems don't seem to present when taking pictures with the [camera2 basic example app](https://github.com/googlesamples/android-Camera2Basic) . I'm not sure if that helps :(

yaraki, 2016-10-18:
Thanks, Iván. I will keep on investigating.
If anyone can provide more information (camera characteristics on those devices, etc), it will be very helpful.

C4Grey, 2016-10-29:
I've encountered the same issue on S7, it may be resolved via rotation of a bitmap according to EXIF rotation:
````kotlin
val file = File(path)
if (file.exists()) {
	val exif = ExifInterface(path)
	val orientation = exif.getAttributeInt(ExifInterface.TAG_ORIENTATION, 1)
	val degrees = when (orientation) {
		ExifInterface.ORIENTATION_NORMAL -> 0
		ExifInterface.ORIENTATION_ROTATE_90 -> 90
		ExifInterface.ORIENTATION_ROTATE_180 -> 180
		ExifInterface.ORIENTATION_ROTATE_270 -> 270
		else -> 0
	}
	BitmapFactory.decodeFile(file, previewSize.maxValue)?.let { it ->
		if (degrees > 0) {
			val matrix = Matrix()
			matrix.postRotate(degrees.toFloat())
			imageView.setImageBitmap(Bitmap.createBitmap(it, 0, 0, it.width, it.height, matrix, true))
		} else {
			imageView.setImageBitmap(it)
		}
	} ?: e("Can't create preview bitmap")
}
````

snowpong, 2017-02-21:
Printing out EXIF orientation on stored images gives 6 for both front and back facing camera on a Sony Xperia Z2. However, the font facing camera has an image that is rotated 180 (upside-down). Tell me what more you need @yaraki

suomi35, 2017-03-01:
I am experiencing a 180 rotation (upside down) on the front camera of a Samsung Galaxy S4. The EXIF reports 8 (90 degrees) when the device is held in portrait position. Please tell me what other info you need @yaraki 
Thanks!

snowpong, 2017-03-08:
@yaraki My Sony Xperia Z2 (API23) falls back to the Camera1 implementation, and it seems fixed by https://github.com/google/cameraview/pull/115 perhaps @WeRockStar could test it as well?

n0m0r3pa1n, 2017-03-09:
@snowpong the Xperia Z2 I have does not fail back to Camera1 and it does not work on Android 7.0 (24). Did you update your own?

snowpong, 2017-03-09:
@n0m0r3pa1n I've just updated the OS on the Z2 whenever a new update came, just like a normal user. If that answers your question?

WeRockStar, 2017-03-20:
Thanks. I fix this problem with the this solution.

```java
int orientation = exif.getAttributeInt(ExifInterface.TAG_ORIENTATION, ExifInterface.ORIENTATION_UNDEFINED);
        switch (orientation) {
            case ExifInterface.ORIENTATION_NORMAL:
                break;
            case ExifInterface.ORIENTATION_FLIP_HORIZONTAL:
                matrix.setScale(-1, 1);
                break;
            case ExifInterface.ORIENTATION_ROTATE_180:
                matrix.setRotate(180);
                break;
            case ExifInterface.ORIENTATION_FLIP_VERTICAL:
                matrix.setRotate(180);
                matrix.postScale(-1, 1);
                break;
            case ExifInterface.ORIENTATION_TRANSPOSE:
                matrix.setRotate(90);
                matrix.postScale(-1, 1);
                break;
            case ExifInterface.ORIENTATION_ROTATE_90:
                matrix.setRotate(90);
                break;
            case ExifInterface.ORIENTATION_TRANSVERSE:
                matrix.setRotate(-90);
                matrix.postScale(-1, 1);
                break;
            case ExifInterface.ORIENTATION_ROTATE_270:
                matrix.setRotate(-90);
                break;
```