google/cameraview #31: Cannot Instantiate CameraView in content_main.xml

shinigami1494, 2016-10-10:
I am getting the following error trace when I try to place the CameraView in the content_main.xml file:
`<com.google.android.cameraview.CameraView
            android:id="@+id/camera"
            android:layout_width="10px"
            android:layout_height="10px"
            android:adjustViewBounds="true"
            app:autoFocus="true"
            app:aspectRatio="4:3"
            app:facing="back"/>`

`java.lang.UnsupportedOperationException: Unsupported Service: camera
	at com.android.layoutlib.bridge.android.BridgeContext.getSystemService(BridgeContext.java:602)
	at com.google.android.cameraview.Camera2.<init>(Camera2.java:191)
	at com.google.android.cameraview.CameraView.<init>(CameraView.java:102)
	at com.google.android.cameraview.CameraView.<init>(CameraView.java:85)
	at sun.reflect.NativeConstructorAccessorImpl.newInstance0(Native Method)
	at sun.reflect.NativeConstructorAccessorImpl.newInstance(NativeConstructorAccessorImpl.java:62)
	at sun.reflect.DelegatingConstructorAccessorImpl.newInstance(DelegatingConstructorAccessorImpl.java:45)
	at java.lang.reflect.Constructor.newInstance(Constructor.java:423)
	at org.jetbrains.android.uipreview.ViewLoader.createNewInstance(ViewLoader.java:465)
	at org.jetbrains.android.uipreview.ViewLoader.loadClass(ViewLoader.java:172)
	at org.jetbrains.android.uipreview.ViewLoader.loadView(ViewLoader.java:105)
	at com.android.tools.idea.rendering.LayoutlibCallbackImpl.loadView(LayoutlibCallbackImpl.java:186)
	at android.view.BridgeInflater.loadCustomView(BridgeInflater.java:334)
	at android.view.BridgeInflater.loadCustomView(BridgeInflater.java:345)
	at android.view.BridgeInflater.createViewFromTag(BridgeInflater.java:245)
	at android.view.LayoutInflater.createViewFromTag(LayoutInflater.java:727)
	at android.view.LayoutInflater.rInflate_Original(LayoutInflater.java:858)
	at android.view.LayoutInflater_Delegate.rInflate(LayoutInflater_Delegate.java:70)
	at android.view.LayoutInflater.rInflate(LayoutInflater.java:834)
	at android.view.LayoutInflater.rInflateChildren(LayoutInflater.java:821)
	at android.view.LayoutInflater.rInflate_Original(LayoutInflater.java:861)
	at android.view.LayoutInflater_Delegate.rInflate(LayoutInflater_Delegate.java:70)
	at android.view.LayoutInflater.rInflate(LayoutInflater.java:834)
	at android.view.LayoutInflater.rInflateChildren(LayoutInflater.java:821)
	at android.view.LayoutInflater_Delegate.parseInclude(LayoutInflater_Delegate.java:197)
	at android.view.LayoutInflater.parseInclude(LayoutInflater.java:902)
	at android.view.LayoutInflater.rInflate_Original(LayoutInflater.java:854)
	at android.view.LayoutInflater_Delegate.rInflate(LayoutInflater_Delegate.java:70)
	at android.view.LayoutInflater.rInflate(LayoutInflater.java:834)
	at android.view.LayoutInflater.rInflateChildren(LayoutInflater.java:821)
	at android.view.LayoutInflater.inflate(LayoutInflater.java:518)
	at android.view.LayoutInflater.inflate(LayoutInflater.java:397)
	at com.android.layoutlib.bridge.impl.RenderSessionImpl.inflate(RenderSessionImpl.java:324)
	at com.android.layoutlib.bridge.Bridge.createSession(Bridge.java:429)
	at com.android.ide.common.rendering.LayoutLibrary.createSession(LayoutLibrary.java:389)
	at com.android.tools.idea.rendering.RenderTask$2.compute(RenderTask.java:548)
	at com.android.tools.idea.rendering.RenderTask$2.compute(RenderTask.java:533)
	at com.intellij.openapi.application.impl.ApplicationImpl.runReadAction(ApplicationImpl.java:966)
	at com.android.tools.idea.rendering.RenderTask.createRenderSession(RenderTask.java:533)
	at com.android.tools.idea.rendering.RenderTask.lambda$inflate$72(RenderTask.java:659)
	at java.util.concurrent.FutureTask.run(FutureTask.java:266)
	at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1142)
	at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:617)
	at java.lang.Thread.run(Thread.java:745)
`

yaraki, 2016-10-14:
Needs more info.

shinigami1494, 2016-10-15:
I am not sure what more information I could provide since I get this error in the layout preview as soon as I place the aforementioned xml in my content_main.xml file.

I am adding my code that uses CameraView, maybe that might help.

yaraki, 2016-10-18:
What is your device? What is the API level of that device?

andreidiaconu, 2017-02-21:
This error appears in Preview in Android Studio (not an an actual device). Perhaps we could use View.isInEditMode() as Android Studio suggests?
> Tip: Use View.isInEditMode() in your custom views to skip code or show sample data when shown in the IDE.