google/cameraview #32: what is this :\

ahmadnajar10, 2016-10-11:
Process: com.google.android.cameraview.demo, PID: 17229
                  java.lang.NullPointerException: Attempt to invoke virtual method 'android.hardware.camera2.CaptureRequest$Builder android.hardware.camera2.CameraDevice.createCaptureRequest(int)' on a null object reference
                      at com.google.android.cameraview.Camera2.captureStillPicture(Camera2.java:562)
                      at com.google.android.cameraview.Camera2.takePicture(Camera2.java:325)
                      at com.google.android.cameraview.CameraView.takePicture(CameraView.java:376)
                      at com.google.android.cameraview.demo.MainActivity$1.onClick(MainActivity.java:91)
                      at android.view.View.performClick(View.java:5280)
                      at android.view.View$PerformClick.run(View.java:21239)
                      at android.os.Handler.handleCallback(Handler.java:739)
                      at android.os.Handler.dispatchMessage(Handler.java:95)
                      at android.os.Looper.loop(Looper.java:234)
                      at android.app.ActivityThread.main(ActivityThread.java:5526)
                      at java.lang.reflect.Method.invoke(Native Method)
                      at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:726)
                      at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:616)

WeRockStar, 2016-10-11:
I have a same problem.

yaraki, 2016-10-14:
Needs more info.

ahmadnajar10, 2016-10-21:
i am using the's view in my application  to shoot pic and when i use the front camera dose not work on marshmallow and stop the application  on lollipop works fine

yaraki, 2016-10-24:
What are the devices?

vsumtsov-gh, 2016-11-01:
Hi,
motorola XT1028
OS Version 
5.1

ahmadnajar10, 2016-11-03:
sony xperia z5

mcassiano, 2017-03-16:
Same here, just after permissions have been granted. Nexus 5, 6.0.1.

``java.lang.RuntimeException: Failure delivering result ResultInfo{who=@android:requestPermissions:android:fragment:0, request=42, result=-1, data=Intent { act=android.content.pm.action.REQUEST_PERMISSIONS (has extras) }}``

``Caused by: java.lang.NullPointerException: Attempt to invoke virtual method 'android.hardware.camera2.CaptureRequest$Builder android.hardware.camera
``

sibelius, 2017-12-10:
check permissions first