google/cameraview #34: Null Pointer Exception when capture session is started

ankitbatra11, 2016-10-15:
I have following exception reported twice from Crashlytics in my app. Attaching device statistics.
[image]

Fatal Exception: java.lang.NullPointerException: Attempt to invoke virtual method 'android.view.Surface android.media.ImageReader.getSurface()' on a null object reference
       at com.google.android.cameraview.Camera2.startCaptureSession(SourceFile:470)
       at com.google.android.cameraview.Camera2.access$200(SourceFile:50)
       at com.google.android.cameraview.Camera2$1.onSurfaceTextureAvailable(SourceFile:70)
       at android.view.TextureView.getHardwareLayer(TextureView.java:370)
       at android.view.View.updateDisplayListIfDirty(View.java:14076)
       at android.view.View.getDisplayList(View.java:14122)
       at android.view.View.draw(View.java:14898)
       at android.view.ViewGroup.drawChild(ViewGroup.java:3407)
       at android.view.ViewGroup.dispatchDraw(ViewGroup.java:3201)
       at android.view.View.updateDisplayListIfDirty(View.java:14094)
       at android.view.View.getDisplayList(View.java:14122)
       at android.view.View.draw(View.java:14898)
       at android.view.ViewGroup.drawChild(ViewGroup.java:3407)
       at android.view.ViewGroup.dispatchDraw(ViewGroup.java:3201)
       at android.view.View.draw(View.java:15177)
       at android.view.View.updateDisplayListIfDirty(View.java:14099)
       at android.view.View.getDisplayList(View.java:14122)
       at android.view.View.draw(View.java:14898)
       at android.view.ViewGroup.drawChild(ViewGroup.java:3407)
       at android.view.ViewGroup.dispatchDraw(ViewGroup.java:3201)
       at android.view.View.updateDisplayListIfDirty(View.java:14094)
       at android.view.View.getDisplayList(View.java:14122)
       at android.view.View.draw(View.java:14898)
       at android.view.ViewGroup.drawChild(ViewGroup.java:3407)
       at android.view.ViewGroup.dispatchDraw(ViewGroup.java:3201)
       at android.view.View.updateDisplayListIfDirty(View.java:14094)
       at android.view.View.getDisplayList(View.java:14122)
       at android.view.View.draw(View.java:14898)
       at android.view.ViewGroup.drawChild(ViewGroup.java:3407)
       at android.view.ViewGroup.dispatchDraw(ViewGroup.java:3201)
       at android.view.View.updateDisplayListIfDirty(View.java:14094)
       at android.view.View.getDisplayList(View.java:14122)
       at android.view.View.draw(View.java:14898)
       at android.view.ViewGroup.drawChild(ViewGroup.java:3407)
       at android.view.ViewGroup.dispatchDraw(ViewGroup.java:3201)
       at android.view.View.updateDisplayListIfDirty(View.java:14094)
       at android.view.View.getDisplayList(View.java:14122)
       at android.view.View.draw(View.java:14898)
       at android.view.ViewGroup.drawChild(ViewGroup.java:3407)
       at android.view.ViewGroup.dispatchDraw(ViewGroup.java:3201)
       at android.view.View.updateDisplayListIfDirty(View.java:14094)
       at android.view.View.getDisplayList(View.java:14122)
       at android.view.View.draw(View.java:14898)
       at android.view.ViewGroup.drawChild(ViewGroup.java:3407)
       at android.view.ViewGroup.dispatchDraw(ViewGroup.java:3201)
       at android.view.View.draw(View.java:15177)
       at android.widget.FrameLayout.draw(FrameLayout.java:592)
       at com.android.internal.policy.impl.PhoneWindow$DecorView.draw(PhoneWindow.java:2695)
       at android.view.View.updateDisplayListIfDirty(View.java:14099)
       at android.view.View.getDisplayList(View.java:14122)
       at android.view.ThreadedRenderer.updateViewTreeDisplayList(ThreadedRenderer.java:266)
       at android.view.ThreadedRenderer.updateRootDisplayList(ThreadedRenderer.java:272)
       at android.view.ThreadedRenderer.draw(ThreadedRenderer.java:311)
       at android.view.ViewRootImpl.draw(ViewRootImpl.java:2531)
       at android.view.ViewRootImpl.performDraw(ViewRootImpl.java:2367)
       at android.view.ViewRootImpl.performTraversals(ViewRootImpl.java:1998)
       at android.view.ViewRootImpl.doTraversal(ViewRootImpl.java:1078)
       at android.view.ViewRootImpl$TraversalRunnable.run(ViewRootImpl.java:5885)
       at android.view.Choreographer$CallbackRecord.run(Choreographer.java:767)
       at android.view.Choreographer.doCallbacks(Choreographer.java:580)
       at android.view.Choreographer.doFrame(Choreographer.java:550)
       at android.view.Choreographer$FrameDisplayEventReceiver.run(Choreographer.java:753)
       at android.os.Handler.handleCallback(Handler.java:739)
       at android.os.Handler.dispatchMessage(Handler.java:95)
       at android.os.Looper.loop(Looper.java:135)
       at android.app.ActivityThread.main(ActivityThread.java:5376)
       at java.lang.reflect.Method.invoke(Method.java)
       at java.lang.reflect.Method.invoke(Method.java:372)
       at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:908)
       at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:703)

dumazy, 2016-10-19:
I'm also having this issue on Android 6.0 and 7.0.
I have a CameraActivity where I start the CameraView in onResume and stop it in onPause. When I call startActivityForResult() and return in onActivityResult(), I start another Activty before going through the lifecycle of CameraActivity. Somehow this causes the issue mentioned above when calling cameraView.stop() in onPause()

dumazy, 2016-10-19:
I made a pull request for a null check that avoids this issue and fixes the problem in the case that I mentioned above

ankitbatra11, 2016-10-19:
Thanks. Will pull it.

In my case, I am using service to host the camera view. User can toggle the camera view which I handle by removing and adding the view through window manager. Every time I add it to window, I call start on the camera view and that's where it fails sometimes.

yaraki, 2016-10-21:
Fixed by the pull request.

closed 2016-10-21 by yaraki

LouisCAD, 2018-01-16:
Which pull request?

richardjbs, 2018-10-11:
@yaraki I am still experimenting this issue on the 3eaeac0 commit (latest version at the moment).