google/cameraview #35: How to add manual focus

aDimple, 2016-10-18:
i need add manual focus when touch screen,  how to do it?

yaraki, 2016-10-21:
This is not yet implemented.

vanniktech, 2016-10-21:
Will it be?

yaraki, 2016-10-21:
That's the plan.
This is one of the reasons why CameraView extends FrameLayout instead of TextureView or SurfaceView. We will be able to add focus indicator.
No ETA, though.

samy-baili, 2016-12-02:
Do you have an idea, when this feature will be available ?

jaksab, 2017-01-02:
This feature will be very useful.

arriolac, 2017-06-15:
I forked the repo and added support for tap to focus; however, it is not fully implemented yet for a pull request (i.e. I haven't added support for `Camera1` and it doesn't take into account different display/sensor orientations). If anyone wants to help out, feel free to fork it: https://github.com/arriolac/cameraview/tree/tap_to_focus

Relates to #150

xiaanming, 2017-06-29:
I hope supportting this funtion in quickly

lin18, 2017-07-14:
https://github.com/google/cameraview/pull/181/commits/47b8a4e493cdb5f1085333577d55b749443047e9
https://github.com/lin18/cameraview

sibelius, 2017-12-10:
check this https://github.com/google/cameraview/pull/218