google/cameraview #38: No camera access lead to collapse，need help.

nanchen2251, 2016-10-21:
Hello, I think I met a big trouble, I'm going to do facial recognition to develop a custom camera, through a variety of ways to solve, so find the CameraView, really very good, I can use this to make a small demo, but into my project always quote exception no camera, I really don't know what's wrong, very helpless, who can help me, thank you

xCatG, 2016-10-27:
Did you add 
```
    <uses-feature android:name="android.hardware.camera" />
    <uses-permission android:name="android.permission.CAMERA" />
```
to your `AndroidManifest.xml` ?

nanchen2251, 2016-10-27:
ok,thank you.
 I was in before adding the permission, but I was found by multiple debug cameras being used, it is not clear, but when I put the application uninstall reshipment is ok. The reason is unknown.I really like your camera.thanks.