google/cameraview #39: java.lang.RuntimeException: setParameters failed

asmallkite, 2016-10-23:
When I click the switch_camera in menu

`E/AndroidRuntime: FATAL EXCEPTION: main
                  Process: cn.yibulz.caviewtest, PID: 10550
                  java.lang.RuntimeException: setParameters failed
                      at android.hardware.Camera.native_setParameters(Native Method)
                      at android.hardware.Camera.setParameters(Camera.java:1671)
                      at com.google.android.cameraview.Camera1.adjustCameraParameters(Camera1.java:323)
                      at com.google.android.cameraview.Camera1.openCamera(Camera1.java:288)
                      at com.google.android.cameraview.Camera1.start(Camera1.java:86)
                      at com.google.android.cameraview.Camera1.setFacing(Camera1.java:136)
                      at com.google.android.cameraview.CameraView.setFacing(CameraView.java:292)
                      at cn.yibulz.caviewtest.MainActivity.onOptionsItemSelected(MainActivity.java:194)
                      at android.app.Activity.onMenuItemSelected(Activity.java:2614)
                      at android.support.v4.app.FragmentActivity.onMenuItemSelected(FragmentActivity.java:406)
                      at android.support.v7.app.AppCompatActivity.onMenuItemSelected(AppCompatActivity.java:195)
                      at android.support.v7.view.WindowCallbackWrapper.onMenuItemSelected(WindowCallbackWrapper.java:103)
                      at android.support.v7.view.WindowCallbackWrapper.onMenuItemSelected(WindowCallbackWrapper.java:103)
                      at android.support.v7.app.ToolbarActionBar$2.onMenuItemClick(ToolbarActionBar.java:69)
                      at android.support.v7.widget.Toolbar$1.onMenuItemClick(Toolbar.java:203)
                      at android.support.v7.widget.ActionMenuView$MenuBuilderCallback.onMenuItemSelected(ActionMenuView.java:762)
                      at android.support.v7.view.menu.MenuBuilder.dispatchMenuItemSelected(MenuBuilder.java:810)
                      at android.support.v7.view.menu.MenuItemImpl.invoke(MenuItemImpl.java:152)
                      at android.support.v7.view.menu.MenuBuilder.performItemAction(MenuBuilder.java:957)
                      at android.support.v7.view.menu.MenuBuilder.performItemAction(MenuBuilder.java:947)
                      at android.support.v7.widget.ActionMenuView.invokeItem(ActionMenuView.java:616)
                      at android.support.v7.view.menu.ActionMenuItemView.onClick(ActionMenuItemView.java:153)
                      at android.view.View.performClick(View.java:4444)
                      at android.view.View$PerformClick.run(View.java:18440)
                      at android.os.Handler.handleCallback(Handler.java:733)
                      at android.os.Handler.dispatchMessage(Handler.java:95)
                      at android.os.Looper.loop(Looper.java:136)
                      at android.app.ActivityThread.main(ActivityThread.java:5016)
                      at java.lang.reflect.Method.invokeNative(Native Method)
                      at java.lang.reflect.Method.invoke(Method.java:515)
                      at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:795)
                      at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:611)
                      at dalvik.system.NativeStart.main(Native Method)
`

yaraki, 2016-10-27:
Does the device have a front-facing camera?

asmallkite, 2016-11-22:
yes

asmallkite, 2016-11-22:
yes

ramdroid, 2017-07-03:
I have the same issue on Nexus 7 (first generation) which only has a front facing camera. Actually it already throws an exception when chooseCamera() doesn't find the back camera and sets mCameraId = INVALID_CAMERA_ID, which is never checked again in the code. When I was setting it to 0 then Camera.open() succeeds but then throws the same exception as above when calling Camera.setParameters()