google/cameraview #41: Preview callback

liberorignanese, 2016-10-26:
it would be helpful if there was a callback for reading preview frame.

Carpetfizz, 2017-01-19:
Any update on this? Is there no way to read frame data from the camera?

xCatG, 2017-03-17:
Hi All,
I'm trying to add support for accessing preview frames, my current thinking is in `CameraViewImpl`, add a interface like
```
interface FrameCallback {
    void onFrame(byte[] data, int format);
}
```
We can then add a `mFrameCallback` so each subclasses can deal with it. In `Camera1`, in `openCamera` we can call `setPreviewCallback` to supply us with the frame; in `Camera2` we add an extra `ImageReader` that does the per-frame reading, and on its callback we can again call `onFrame()` with the byte from the `Image` object, as well as its format. 

`int format` in the `onFrame()` call might be changed to a different object containing image info, not sure yet as I haven't start coding.

Where I am not sure is how to accept a external hook from `CameraView` itself. I am not sure how many people will want this behavior by default, so a separate `CameraViewWithPreviewCall` that extends `CameraView` might be more desirable? Also, given these are usually inflated from layout xml, the entry where we set callbacks might not be easy (to be seen, again, I haven't start coding yet).

Any thoughts, comments?

xCatG, 2017-03-21:
Another approach is 
Add `void onFrame(byte[] data, int format);' to `CameraView.Callback`, as well as `CameraViewImpl.Callback`. In `Camera1`, we call this in the `onPreviewFrame()`; in `Camera2`, this is called on a `ImageReader`'s `OnImageAvailableListener`.

@yaraki out of the two approaches, which one you prefer? Or you have other approaches that I should explore more about?

xCatG, 2017-03-21:
The reason for including `format` as part of the callback is because `Camera1` uses `ImageFormat.NV21` or `ImageFormat.YV12` and `Camera2` recommends `ImageFormat.YUV_420_888`. 

Haven't dig more, but we might have to also include image width and height, event stride value to correctly decode things. Still trying things out right now.

xCatG, 2017-03-22:
PR #123 by @Sumsar is basically the second approach.

jackyhieu1211, 2017-05-17:
@xCatG 
Camera1 on device HTC Butterfly 4.3 API 18 not call
mCamera.setPreviewCallback(new Camera.PreviewCallback() {
@override
public void onPreviewFrame(byte[] data, Camera camera) {
Timber.e("onPreviewFrame");
final Camera.Parameters parameters = camera.getParameters();
final int width = parameters.getPreviewSize().width;
final int height = parameters.getPreviewSize().height;
mCallback.onPreviewFrame(data, parameters.getPreviewFormat(), width, height);
}
});