google/cameraview #48: Preview doesn't show whole picture

C4Grey, 2016-11-11:
I've build a fragment containing a CameraView instance and a toolbar, just like a default Android Camera app. I am trying to achive a smooth transition between a preview and a taken photo, but on some devices preview showing only a part of a picture, while resulting image is looking normal. I've tried to tweak the adjustViewBounds parameter, but it changes nothing. 
P.S.
I've figured out that changing of a width and height according to an optimal size value [here](https://github.com/google/cameraview/blob/d0b508a5ea4fff2bbb3313fe86216980ed08aac0/library/src/main/java/com/google/android/cameraview/CameraView.java#L177) may help, but a preview image is not centered according to its frame.

athornz, 2017-05-06:
I've got the same issue. The preview from the CameraView is cropped or zoomed, but the picture taken is full size

stonecompass, 2017-06-22:
I'm having the same issue.

itaybia, 2017-08-03:
seeing same issue. can't use the lib...

iLeafSolutionsPvtLtd, 2019-05-07:
Anybody found any solution for this? I too facing similar issue while catching with front cam.