google/cameraview #51: java.lang.RuntimeException: autoFocus failed

RahulRaktale, 2016-11-21:
On tapping multiple times on the capture image button the demo app crashes.
The issue appeared on Samsung GT-I9300I running Android 4.4.4
(By the way very helpful library. Can't even begin to tell how big a help this has been.)

com.google.android.cameraview.demo E/AndroidRuntime: FATAL EXCEPTION: main
                                                                                    Process: com.google.android.cameraview.demo, PID: 22695
                                                                                    java.lang.RuntimeException: autoFocus failed
                                                                                        at android.hardware.Camera.native_autoFocus(Native Method)
                                                                                        at android.hardware.Camera.autoFocus(Camera.java:1131)
                                                                                        at com.google.android.cameraview.Camera1.takePicture(Camera1.java:212)
                                                                                        at com.google.android.cameraview.CameraView.takePicture(CameraView.java:378)
                                                                                        at com.google.android.cameraview.demo.MainActivity$1.onClick(MainActivity.java:91)
                                                                                        at android.view.View.performClick(View.java:4741)
                                                                                        at android.view.View$PerformClick.run(View.java:19384)
                                                                                        at android.os.Handler.handleCallback(Handler.java:733)
                                                                                        at android.os.Handler.dispatchMessage(Handler.java:95)
                                                                                        at android.os.Looper.loop(Looper.java:146)
                                                                                        at android.app.ActivityThread.main(ActivityThread.java:5679)
                                                                                        at java.lang.reflect.Method.invokeNative(Native Method)
                                                                                        at java.lang.reflect.Method.invoke(Method.java:515)
                                                                                        at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:1291)
                                                                                        at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:1107)
                                                                                        at dalvik.system.NativeStart.main(Native Method)

JakeWoki, 2017-01-13:
Device: Honor SCL-AL00
OS: EMUI3.1.1(Android 5.1.1)

olivercm, 2017-03-13:
And OPPO R7

xiaanming, 2017-10-10:
The same issue on  vivo X3L

zhaoyangxihui, 2019-11-14:
Android 7.1

FitApps7, 2020-07-03:
Redmi 8 
Android version 9