google/cameraview #53: Failed to start camera session

jaksab, 2016-11-26:
I try to start camera after user set permission

@Override
    public void onRequestPermissionsResult(int requestCode, @NonNull String[] permissions,
                                           @NonNull int[] grantResults) {
        switch (requestCode) {
            case REQUEST_CAMERA_PERMISSION:
                if (grantResults.length > 0
                        && grantResults[0] == PackageManager.PERMISSION_GRANTED) {
                    cameraView.start();
                }
                break;
        }
    }

but I get exception:

FATAL EXCEPTION: main
                                                                           Process: ua.com.oncreate.infitting, PID: 15583
                                                                           java.lang.RuntimeException: Failed to start camera session
                                                                               at com.google.android.cameraview.Camera2.startCaptureSession(Camera2.java:443)
                                                                               at com.google.android.cameraview.Camera2$1.onOpened(Camera2.java:65)
                                                                               at android.hardware.camera2.impl.CameraDeviceImpl$1.run(CameraDeviceImpl.java:134)
                                                                               at android.os.Handler.handleCallback(Handler.java:739)
                                                                               at android.os.Handler.dispatchMessage(Handler.java:95)
                                                                               at android.os.Looper.loop(Looper.java:158)
                                                                               at android.app.ActivityThread.main(ActivityThread.java:7225)
                                                                               at java.lang.reflect.Method.invoke(Native Method)
                                                                               at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:1230)
                                                                               at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:1120)

lukaspili, 2017-01-07:
Same on nexus 6p, android 7.1.1.
Right after authorising camera permission:
```java.lang.RuntimeException: Failed to start camera session```

Next calls work properly.

protocol10, 2017-01-23:
It appeared on Nexus 5x for android 7.1.1 , after granting permission for Camera
`java.lang.RuntimeException: Failed to start camera session
                                                                     at com.google.android.cameraview.Camera2.startCaptureSession(Camera2.java:462)
                                                                     at com.google.android.cameraview.Camera2$1.onOpened(Camera2.java:64)
                                                                     at android.hardware.camera2.impl.CameraDeviceImpl$1.run(CameraDeviceImpl.java:134)
                                                                     at android.os.Handler.handleCallback(Handler.java:751)
                                                                     at android.os.Handler.dispatchMessage(Handler.java:95)
                                                                     at android.os.Looper.loop(Looper.java:154)
                                                                     at android.app.ActivityThread.main(ActivityThread.java:6119)
                                                                     at java.lang.reflect.Method.invoke(Native Method)
                                                                     at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:886)
                                                                     at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:776)`

xiaoyunfei, 2017-08-01:
It appeared on LGE Nexus 5x for android O，API25 , after granting permission for Camera

 java.lang.RuntimeException: Failed to start camera session
at com.google.android.cameraview.Camera2.startCaptureSession(Camera2.java:486)
at com.google.android.cameraview.Camera2$1.onOpened(Camera2.java:75)
at android.hardware.camera2.impl.CameraDeviceImpl$1.run(CameraDeviceImpl.java:139)
at android.os.Handler.handleCallback(Handler.java:769)
at android.os.Handler.dispatchMessage(Handler.java:98)
at android.os.Looper.loop(Looper.java:164)
at android.app.ActivityThread.main(ActivityThread.java:6535)
at java.lang.reflect.Method.invoke(Native Method)
at com.android.internal.os.Zygote$MethodAndArgsCaller.run(Zygote.java:240)
at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:767)

yanxinmiao, 2017-12-28:
i got this issue to on HuaWei Phone ， but only the fisrt time to use camera 。

who know how to solve this bug ？

JeetSingh4, 2017-12-29:
I solved it by giving a delay of 500 mS while creating capture session because the camera configurations are not fully loaded until then, hence this delay.

```
void startCaptureSession() {
        if (!isCameraOpened() || !mPreview.isReady() || mImageReader == null) {
            return;
        }
        previewSize = chooseOptimalSize();
        mPreview.setBufferSize(previewSize.getWidth(), previewSize.getHeight());
        new Handler().postDelayed(new Runnable() {
            @Override
            public void run() {
                try {
                    Surface surface = mPreview.getSurface();
                    mPreviewRequestBuilder = mCamera.createCaptureRequest(CameraDevice.TEMPLATE_PREVIEW);
                    mPreviewRequestBuilder.addTarget(surface);
                    mCamera.createCaptureSession(Arrays.asList(surface, mImageReader.getSurface()),
                            mSessionCallback, mBackgroundHandler);
                } catch (CameraAccessException e) {
                    throw new RuntimeException("Failed to start camera session");
                }
            }
        }, 500);
    }

yanxinmiao, 2018-01-02:
@JeetSingh4 Thanks 
 Your solution solved my issue .

zweinz, 2018-01-08:
Calling cameraView.start() on a delay also solved this issue for me but now I'm running into #11 instead.

JeetSingh4, 2018-01-09:
@zweinz  buddy I hope this helps [https://stackoverflow.com/questions/40146485/session-has-been-closed-further-changes-are-illegal/48019131#48019131](url)

srikanthnama, 2018-02-28:
I am still facing issue even after using your solution

Seckawijoki, 2019-01-24:
@JeetSingh4  Crash that the mCamera will sometimes become null will happen in some devices like Mi and Huawei, so it need a prerequisite sentence before calling createCaptureRequest inside Runnable.