google/cameraview #54: Saves only highest-resolution image?

smhk, 2016-11-26:
Thanks for this incredible library.  When you save, it seems to save the highest-available resolution matching the ratio.  Often you want mid-resolution.  (1) where to look in the code to change this?  (2) Perhaps in the future it would be possible to add a tag app:save_resolution="high" "mid" "low".  Thanks again.

jaksab, 2016-11-30:
Or set the some resolution from available camera parameters

owlstack, 2016-12-14:
How is the camera using the highest resolution image right now?

hujiaweibujidao, 2017-09-26:
@smhk 

for Camera1, the code lies in line 333

```
final Size pictureSize = mPictureSizes.sizes(mAspectRatio).last();
```

for Camera2, the code lies in line 450

```
Size largest = mPictureSizes.sizes(mAspectRatio).last();
```