google/cameraview #59: Result File is Strange

FongMi, 2016-12-06:
This is Preview in App.
[image]

This is Result File.
[image]

Why the result picture is dark ??

yaraki, 2016-12-07:
Nice keyboard. I suppose this is a device-dependent issue with the flash.

DinoMC, 2017-06-01:
I have the same issue on a Sony Xperia Z3. I don't have a picture of the preview, but here's the result file : 
[image]

This is in daylight, with good lighting, and the flash didn't trigger (which is normal in those conditions). Could this still be an issue with the flash?

It works fine on my other devices.

closed 2018-10-01 by FongMi