google/cameraview #60: How to import this library into project?

mladenrakonjac, 2016-12-06:
Couse there is no jcenter, how I can import this?

PavelSynek, 2016-12-06:
https://jitpack.io/#google/cameraview

mladenrakonjac, 2016-12-06:
I get this error :
Gradle sync failed: Could not find com.github.google:cameraview:-SNAPSHOT

PavelSynek, 2016-12-06:
I wouldn't use SNAPSHOT, it might lead to unpredictable builds. `compile 'com.github.google:cameraview:d0b508a5ea'` works for me.

mladenrakonjac, 2016-12-06:
I got same error

PavelSynek, 2016-12-06:
Oh you also need to add `maven { url 'https://jitpack.io' }` to the project-level `build.gradle`

yaraki, 2016-12-07:
#10

closed 2016-12-07 by yaraki

dbarretto, 2017-05-08:
If you are having issues importing this library please check this link below:

http://stackoverflow.com/questions/32718820/failed-to-resolve-com-github-philjaympandroidchartv2-1-4/37892027#37892027