google/cameraview #61: java.lang.RuntimeException: Unable to resume activity: java.lang.SecurityException: Lacking privileges to access camera service

Saklhc, 2016-12-07:
java.lang.RuntimeException: Unable to resume activity: java.lang.SecurityException: Lacking privileges to access camera service

Saklhc, 2016-12-07:
java.lang.RuntimeException: Unable to resume activity: java.lang.SecurityException: Lacking privileges to access camera service

Saklhc, 2016-12-07:
java.lang.RuntimeException: Unable to resume activity: java.lang.SecurityException: Lacking privileges to access camera service

The camera has authorized

API 22  ,VERSION ANDROID 5.1

yaraki, 2016-12-13:
I can't reproduce this. Are you using the component in your own app? If so, how do you build the app? Does your APK have the camera permissions?

jf-rce, 2016-12-14:
This issue is present in Google's Camera2Basic sample as well:

https://github.com/googlesamples/android-Camera2Basic/issues/42

I am experiencing something similar with a fork of this library ONLY while using Android 6.0. I cannot replicate it using any other version.

yaraki, 2016-12-15:
Hmm, I tried to reproduce this on my Nexus 5 running 6.0.1, but I still cannot reproduce. I tried uninstall & install, revoking camera permission, changing orientation, and switching from other camera apps.
Do you have suggestions for reproducing this?

jf-rce, 2016-12-15:
Yes, I have some suggestions. I was able to remedy the problem with some workflow changes in my app. The issue is still present in the library, but it is avoidable.

I'm only able to cause the crash to occur in one of five apps in which I'm using a fork of your library. In the app in which the crash occurs, it happens when I open an AlertDialog while the camera is closing (or about to be closed), then hit the back button while the dialog is open to close the dialog and restart the camera. Something like this will crash ONLY on Android 6.0:

```
mCameraView.stop();
AlertDialog.Builder builder = new AlertDialog.Builder(mContext);
builder.setMessage("blah")
       .setOnDismissListener(new OnDismissListener() {
          @Override
         public void onDismiss(DialogInterface dialog) {
              mCameraView.start();
          }
        })
       .create()
       .show();
```

You can get around it by respecting the CameraView's callback methods and only opening the dialog AFTER the camera has closed. This code is safe:

```
// Stop the camera
mCameraView.stop()

private CameraView.Callback mCallback = new CameraView.Callback() {

      ...

      @Override
      public void onCameraClosed(CameraView cameraView) {
        // Wait until camera is fully closed to open dialog
         AlertDialog.Builder builder = new AlertDialog.Builder(mContext);
         builder.setMessage("blah")
                .setOnDismissListener(new OnDismissListener() {
                   @Override
                  public void onDismiss(DialogInterface dialog) {
                       mCameraView.start();
                   }
                 })
                .create()
                .show();
      }

      ...
};
```

@p4block mentions that he thinks the problem is related to alert themes in this thread:

https://github.com/googlesamples/android-Camera2Basic/issues/42

I might be inclined to agree. 

Hope this helps.