google/cameraview #63: Camera3 Support failing

BijoySingh, 2016-12-10:
Getting a black screen on my Pixel device. Logcat shows this error:

```
12-10 09:51:35.518   570  1434 E mm-camera: <C2D   ><ERROR> 394: c2d_module_start_session: Cannot open ION
12-10 09:51:35.521   570  1441 E mm-camera: <STATS_AIS ><ERROR> 605: sensor_native_thread_handler: Laser Sensor not found!
12-10 09:51:35.548   570 30919 E mm-camera: <MCT   ><ERROR> 537: mct_pipeline_lookup_session_data: Couldn't find stream for sessionid 15
12-10 09:51:35.548   570 30919 E mm-camera: <CPP   ><ERROR> 2206: cpp_module_notify_add_stream: failed, CPP current version can not support rotation
12-10 09:51:35.576   570   570 E Camera2-Metadata: Mismatched tag type when updating entry enable (-2146893824) of type byte; got type int32 data instead
12-10 09:51:35.576   570   570 E Camera2-Metadata: Mismatched tag type when updating entry is_main (-2146893823) of type byte; got type int32 data instead
12-10 09:51:35.578   570  9921 E QCamera : <MCI><ERROR> mm_channel_fsm_fn_stopped: 893: invalid state (1) for evt (6)
12-10 09:51:35.578   570  9921 E QCamera3HWI: int qcamera::QCamera3HardwareInterface::configureStreamsPerfLocked(camera3_stream_configuration_t *): Blob size greater than 4k and multiple streams are on encoder output
12-10 09:51:35.578   570  9921 E Camera3-Device: Camera 0: configureStreamsLocked: Set of requested inputs/outputs not supported by HAL
12-10 09:51:35.578   570  9921 E CameraDeviceClient: endConfigure: Camera 0: Unsupported set of inputs/outputs provided
12-10 09:51:35.579 32440 32440 E CameraCaptureSession: Session 0: Failed to create capture session; configuration failed
12-10 09:51:35.581   570 30919 E QCamera : <MCI><ERROR> mm_channel_fsm_fn_stopped: 893: invalid state (1) for evt (6)
12-10 09:51:35.582   570 30919 E QCamera3HWI: int qcamera::QCamera3HardwareInterface::configureStreamsPerfLocked(camera3_stream_configuration_t *): Blob size greater than 4k and multiple streams are on encoder output
12-10 09:51:35.582   570 30919 E Camera3-Device: Camera 0: configureStreamsLocked: Set of requested inputs/outputs not supported by HAL
12-10 09:51:35.582   570 30919 E CameraDeviceClient: endConfigure: Camera 0: Unsupported set of inputs/outputs provided
12-10 09:51:35.582 32440 32440 E CameraCaptureSession: Session 1: Failed to create capture session; configuration failed
12-10 09:51:35.599 32440 32440 E Camera2 : Failed to configure capture session.
12-10 09:51:35.599 32440 32440 E Camera2 : Failed to configure capture session.
12-10 09:51:35.634   570  1420 E mm-camera: <SENSOR><ERROR> 37: af_actuator_power_up: failed: rc = -1
12-10 09:51:35.634   570  1420 E mm-camera: <SENSOR><ERROR> 688: af_actuator_init: failed rc -5
12-10 09:51:35.957   570  9921 E QCamera : <MCI><ERROR> mm_channel_fsm_fn_stopped: 893: invalid state (1) for evt (6)
12-10 09:51:35.957   570  9921 E QCamera3HWI: int qcamera::QCamera3HardwareInterface::configureStreamsPerfLocked(camera3_stream_configuration_t *): Blob size greater than 4k and multiple streams are on encoder output
12-10 09:51:35.957   570  9921 E Camera3-Device: Camera 0: configureStreamsLocked: Set of requested inputs/outputs not supported by HAL
12-10 09:51:35.958   570  9921 E CameraDeviceClient: endConfigure: Camera 0: Unsupported set of inputs/outputs provided
12-10 09:51:35.959 32440 32440 E CameraCaptureSession: Session 2: Failed to create capture session; configuration failed
12-10 09:51:35.960 32440 32440 E Camera2 : Failed to configure capture session.
12-10 09:51:40.808  2205  2213 E DataBuffer: Internal data leak within a DataBuffer object detected!  Be sure to explicitly call release() on all DataBuffer extending objects when you are done with them. (internal object: com.google.android.gms.common.data.DataHolder@c2c7b1b)
12-10 09:51:40.809  2205  2213 E DataBuffer: Internal data leak within a DataBuffer object detected!  Be sure to explicitly call release() on all DataBuffer extending objects when you are done with them. (internal object: com.google.android.gms.common.data.DataHolder@e54d4b8)
12-10 09:51:40.809  2205  2213 E DataBuffer: Internal data leak within a DataBuffer object detected!  Be sure to explicitly call release() on all DataBuffer extending objects when you are done with them. (internal object: com.google.android.gms.common.data.DataHolder@13ed291)
```

yaraki, 2016-12-13:
I can't reproduce this on my Pixel running 7.1.1. What is your build number?

shipra578, 2017-02-24:
remove "ImageFormat.JPEG" support.
e.g. remove below line if you are using.
 ImageReader.newInstance(width, height, ImageFormat.JPEG, 1);

Use RAW-SENSOR:
 ImageReader mRawImageReader = ImageReader.newInstance(rawWidth, rawHeight, ImageFormat.RAW_SENSOR, 1);

ShreyashPromact, 2017-03-01:
@BijoySingh 
Do you have Nougat OS in your pixel device?

Sincerely,
Shreyash

ShreyashPromact, 2017-03-01:
@yaraki 

Can you please try making full screen as Camera view and check.

I have something similar issue here: #98 

Sincerely,
Shreyash

closed 2017-03-13 by yaraki

anurag-176, 2018-10-13:
@shipra578
Thanks, that works!

PR #117 (2017-03-13): Fix preview size issue. It is now below or equal to the 1080p Camera2 guarantee
Taking maximum preview size example from [Camera2Basic](https://github.com/googlesamples/android-Camera2Basic/blob/master/Application/src/main/java/com/example/android/camera2basic/Camera2BasicFragment.java#L384)

The motivation is from the [official docs](https://developer.android.com/reference/android/hardware/camera2/CameraDevice.html) where it states that 
> For the maximum size column, PREVIEW refers to the best size match to the device's screen resolution, or to 1080p (1920x1080), whichever is smaller

This m [... cut]