google/cameraview #64: Bag after update library

jaksab, 2016-12-13:
Hi!
After last update: f0eef9ade1969f9dafef7138172468e409c3210c I get next bug (not olways, but offen):

FATAL EXCEPTION: main
 Process: ua.com.oncreate.infitting, PID: 6792
java.lang.RuntimeException: takePicture failed
at android.hardware.Camera.native_takePicture(Native Method)
at android.hardware.Camera.takePicture(Camera.java:1523)
at com.google.android.cameraview.Camera1.takePictureInternal(Camera1.java:227)
at com.google.android.cameraview.Camera1.takePicture(Camera1.java:222)
at com.google.android.cameraview.CameraView.takePicture(CameraView.java:388)
at ua.com.oncreate.infitting.activity.camera.CameraActivity.onClick(CameraActivity.java:264)
at android.view.View.performClick(View.java:5697)
 at android.view.View$PerformClick.run(View.java:22526)
at android.os.Handler.handleCallback(Handler.java:739)
at android.os.Handler.dispatchMessage(Handler.java:95)
at android.os.Looper.loop(Looper.java:158)
at android.app.ActivityThread.main(ActivityThread.java:7225)
at java.lang.reflect.Method.invoke(Native Method)
at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:1230)
at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:1120)

I run project on Note 4 (6.0.1), but library used camera1 API, why? What I should do with this bug?

yaraki, 2016-12-15:
We started using Camera1 API when Camera2 is in legacy mode. The change fixed issues on some devices, but broke on other.
I don't have Note 4, but I'll try to reproduce this with some other device.

sibelius, 2017-12-12:
@jaksab is this still happening?