google/cameraview #66: question about autofocusing.

pastafarianGit, 2016-12-14:
hey,
When I try to implement this library, autofocus doesn't work. I basically copied MainActivity from git into my project. Only thing i added to that is  app:autoFocus="true" to CameraView. But it's still won't autofocus even though when i installed demoapp apk on same device it works perfectly.

Check difference in this link.
[](https://docs.google.com/drawings/d/1N04UGXC89FcZOysQLsLHJ8oANKEZlCZqIDcJ8F8TJC0/edit?usp=sharing)

Any tips would be greatly appreciated!

yaraki, 2016-12-15:
Auto focus should be on by default. Can you share other part of your code?

pastafarianGit, 2016-12-15:
My bad, I don't think it was problem with autofocus. Problem probably lies in parameters for previewBuilder.
CaptureRequest.Builder mPreviewRequestBuilder;
mPreviewRequestBuilder.set();

You can remove this question. Thanks for time!

closed 2016-12-20 by yaraki