google/cameraview #73: Samsung s4 and htc app crash while capturing picture

sb-hitesh, 2016-12-22:
Exception java.lang.NullPointerException: Attempt to invoke virtual method 'int android.hardware.camera2.CameraCaptureSession.capture(android.hardware.camera2.CaptureRequest, android.hardware.camera2.CameraCaptureSession$CaptureCallback, android.os.Handler)' on a null object reference
com.google.android.cameraview.Camera2.unlockFocus (Camera2.java:617)
com.google.android.cameraview.Camera2$6.onCaptureCompleted (Camera2.java:601)
java.lang.reflect.Method.invoke (Method.java)
java.lang.reflect.Method.invoke (Method.java:372)
android.hardware.camera2.dispatch.InvokeDispatcher.dispatch (InvokeDispatcher.java:39)
android.hardware.camera2.dispatch.HandlerDispatcher$1.run (HandlerDispatcher.java:65)
android.os.Handler.handleCallback (Handler.java:739)
android.os.Handler.dispatchMessage (Handler.java:95)
android.os.Looper.loop (Looper.java:145)
android.app.ActivityThread.main (ActivityThread.java:5951)
java.lang.reflect.Method.invoke (Method.java)
java.lang.reflect.Method.invoke (Method.java:372)
com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run (ZygoteInit.java:1400)
com.android.internal.os.ZygoteInit.main (ZygoteInit.java:1195)

nihar4154, 2017-12-28:
same crash on Moto g5 plus

Lans, 2018-09-14:
how to solve it?