google/cameraview #80: takePicture failed

JakeWoki, 2017-01-13:
``` Java
java.lang.RuntimeException: takePicture failed
at android.hardware.Camera.native_takePicture(Native Method)
at android.hardware.Camera.takePicture(Camera.java:1561)
at com.google.android.cameraview.Camera1.takePictureInternal(Camera1.java:230)
at com.google.android.cameraview.Camera1$2.onAutoFocus(Camera1.java:221)
at android.hardware.Camera$EventHandler.handleMessage(Camera.java:1199)
at android.os.Handler.dispatchMessage(Handler.java:102)
at android.os.Looper.loop(Looper.java:135)
at android.app.ActivityThread.main(ActivityThread.java:5593)
at java.lang.reflect.Method.invoke(Native Method)
at java.lang.reflect.Method.invoke(Method.java:372)
at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:967)
at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:762)
```
Device: Honor SCL-AL00
OS: EMUI3.1.1(Android 5.1.1)

bzy601638015, 2017-01-19:
＋1

anuragopenxcell, 2017-01-20:
+1, it always ends up in Camera1 irrespective of the API version

closed 2017-01-24 by yaraki

deepikaopenxcell, 2017-01-27:
I am still getting this issue.
it is random please help me.

krupalagile, 2017-02-14:
still getting issue

dannyroa, 2017-02-21:
@yaraki: I have this issue as well.

cwliu, 2017-05-20:
Getting this issue, here is the traceback:
 
[image]

aswalalka, 2017-07-05:
Is it fixed, I am facing the same issue

commit 475546b6 (2017-01-24): Fix auto focus on Camera1

Fix auto focus on some devices that run with Camera1 API.

Resolves #75
Resolves #80

Change-Id: I83d2b2d2c76f09373fc2efe4a6161a2863c90d5c