google/cameraview #84: Front Camera issue

rajeshct, 2017-01-21:
Image rotated 180 degree while capturing  from front camera

Android version:7.1
Model: Yureka

n0m0r3pa1n, 2017-01-22:
Yup, this is normal on Android. You have to implement a method to analyze and rotate the image:
`
    // isOldCameraApi is a flag I use if the Preview is Camera1 or Camera2 because on older devices
   // you just have to rotate the image on certain types
    public static Bitmap correctImageOrientation(File file, boolean isOldCameraApi) {
        BitmapFactory.Options options = new BitmapFactory.Options();
        Bitmap bitmap = null;
        try {
            bitmap = BitmapFactory.decodeStream(new FileInputStream(file), null, options);
            options.inPreferredConfig = Bitmap.Config.ARGB_8888;
            bitmap = BitmapFactory.decodeStream(new FileInputStream(file), null, options);
            ExifInterface ei = new ExifInterface(file.getAbsolutePath());
            int orientation = ei.getAttributeInt(ExifInterface.TAG_ORIENTATION,
                    ExifInterface.ORIENTATION_UNDEFINED);

            return getRotatedBitmap(bitmap, orientation, isOldCameraApi);

        } catch (IOException e) {
            e.printStackTrace();
            return bitmap;
        }
    }

    private static Bitmap getRotatedBitmap(Bitmap bitmap, int orientation, boolean isOldCameraApi) {
        Timber.d("Orientation: " + orientation + " OldCameraAPI: " + isOldCameraApi);
        switch (orientation) {
            case ExifInterface.ORIENTATION_ROTATE_90:
                bitmap = rotateImage(bitmap, 90);
                break;
            case ExifInterface.ORIENTATION_ROTATE_180:
                bitmap = rotateImage(bitmap, 180);
                break;
            case ExifInterface.ORIENTATION_ROTATE_270:
                bitmap = rotateImage(bitmap, 270);
                break;
            case ExifInterface.ORIENTATION_NORMAL:
            case ExifInterface.ORIENTATION_UNDEFINED:
                if (isOldCameraApi) {
                    bitmap = rotateImage(bitmap, 180);
                }
                break;
            default:
                break;
        }
        return bitmap;
    }

    public static Bitmap rotateImage(Bitmap source, float angle) {
        Matrix matrix = new Matrix();
        matrix.postRotate(angle);
        Bitmap bitmap = Bitmap.createBitmap(source, 0, 0, source.getWidth(), source.getHeight(), matrix,
                true);

        return bitmap;
    }`

yaraki, 2017-01-23:
The library should provide the right orientation on any device. I don't have Yureka with me, so I'd really appreciate someone with these devices to contribute.

protocol10, 2017-01-25:
Same appeared on Asus zenphone for rear camera

deepikaopenxcell, 2017-01-25:
same appeared in samsung note3 for front camera

rajeshct, 2017-01-26:
I have made some changes in project please have a look their are some changes in project file:

1) Fix crash in android 6.0
2) Front camera image rotation issue
3) Camera opened onResume (Bug in sample code) 

```
/*
 * Copyright (C) 2016 The Android Open Source Project
 *
 * Licensed under the Apache License, Version 2.0 (the "License");
 * you may not use this file except in compliance with the License.
 * You may obtain a copy of the License at
 *
 *      http://www.apache.org/licenses/LICENSE-2.0
 *
 * Unless required by applicable law or agreed to in writing, software
 * distributed under the License is distributed on an "AS IS" BASIS,
 * WITHOUT WARRANTIES OR CONDITIONS OF ANY KIND, either express or implied.
 * See the License for the specific language governing permissions and
 * limitations under the License.
 */

package com.google.android.cameraview.demo;

import android.Manifest;
import android.app.Dialog;
import android.content.DialogInterface;
import android.content.pm.PackageManager;
import android.graphics.Bitmap;
import android.os.Build;
import android.os.Bundle;
import android.os.Environment;
import android.os.Handler;
import android.os.HandlerThread;
import android.support.annotation.NonNull;
import android.support.annotation.StringRes;
import android.support.design.widget.FloatingActionButton;
import android.support.v4.app.ActivityCompat;
import android.support.v4.app.DialogFragment;
import android.support.v4.content.ContextCompat;
import android.support.v7.app.ActionBar;
import android.support.v7.app.AlertDialog;
import android.support.v7.app.AppCompatActivity;
import android.support.v7.widget.Toolbar;
import android.util.Log;
import android.view.Menu;
import android.view.MenuItem;
import android.view.View;
import android.view.ViewGroup;
import android.widget.FrameLayout;
import android.widget.ImageView;
import android.widget.Toast;

import com.google.android.cameraview.AspectRatio;
import com.google.android.cameraview.CameraView;

import java.io.File;
import java.io.FileOutputStream;
import java.io.IOException;
import java.io.OutputStream;
import java.util.Set;

/**
 * This demo app saves the taken picture to a constant file.
 * $ adb pull /sdcard/Android/data/com.google.android.cameraview.demo/files/Pictures/picture.jpg
 */
public class MainActivity extends AppCompatActivity implements
        ActivityCompat.OnRequestPermissionsResultCallback,
        AspectRatioFragment.Listener {
[... 399 lines cut ...]
        matrix.postRotate(angle);
        return Bitmap.createBitmap(source, 0, 0, source.getWidth(), source.getHeight(), matrix, true);
    }
}
```

<?xml version="1.0" encoding="utf-8"?>

<FrameLayout
    xmlns:android="http://schemas.android.com/apk/res/android"
    android:layout_width="match_parent"
    android:layout_height="match_parent">

    <FrameLayout
        android:id="@+id/camera"
        android:layout_width="match_parent"
        android:layout_height="match_parent"
        />

    <android.support.v7.widget.Toolbar
        android:id="@+id/toolbar"
        android:layout_width="match_parent"
        android:layout_height="?attr/actionBarSize"/>

</FrameLayout>

xp-vit, 2017-01-30:
I think that lirary should not rotate image itself, instead it should provide clear info on the fact that image should be rotated. For back camera I was able to manage rotation based on Exif info (relying on  ExifInterface.TAG_ORIENTATION) and then rotating image based on found value.
But I'm having issues with detecting rotation angle for front camera for some devices. I'm also using same logic based on Exif. Here is result of my testings:

Phone | Android version | ExifInterface.TAG_ORIENTATION | actual result
 ----- |----- | ----- |----- 
LG Nexus 4 | 5.1.1 | ExifInterface.ORIENTATION_UNDEFINED = 0 | image rotated 180
LG Nexus 5x | 7.1.1 | ExifInterface.ORIENTATION_UNDEFINED = 0 | image has normal orientation 
HTC Desire 320 | 4.4.2 | ExifInterface.ORIENTATION_NORMAL = 1 | image rotated 180 

I also spend more time and tried same phones with different implementations of Camera in CameraView.java: `Camera1` `Camera2` and `Camera2Api23`. Here are the results:

**LG Nexus 5x, Android 7.1.1 **

Implementation | ExifInterface.TAG_ORIENTATION | actual result
----- | ----- | -----
`Camera1`  | ExifInterface.ORIENTATION_UNDEFINED = 0 | image rotated 180
`Camera2` | ExifInterface.ORIENTATION_UNDEFINED = 0 | image has normal orientation 
`Camera2Api23` | ExifInterface.ORIENTATION_UNDEFINED = 0 |HTC Desire 320 | 4.4.2

**LG Nexus 4, Android 5.0.1 **

Implementation | ExifInterface.TAG_ORIENTATION | actual result
----- | ----- | -----
`Camera1`  | ExifInterface.ORIENTATION_UNDEFINED = 0 | image rotated 180
`Camera2` | ExifInterface.ORIENTATION_UNDEFINED = 0 | image rotated 180
`Camera2Api23` | ExifInterface.ORIENTATION_UNDEFINED = 0 | image rotated 180

I also have testes with Nexus 5, and the result was: image has normal orientation with default lib start, I don't have it handy to provide more details for now.

jinujayakumar, 2017-02-16:
@xp-vit  i am facing the same problem

I found this code from **_Camera2Basic_** and add **_Camera2_** with these code

`private static final SparseIntArray ORIENTATIONS = new SparseIntArray();
    static {
        ORIENTATIONS.append(Surface.ROTATION_0, 90);
        ORIENTATIONS.append(Surface.ROTATION_90, 0);
        ORIENTATIONS.append(Surface.ROTATION_180, 270);
        ORIENTATIONS.append(Surface.ROTATION_270, 180);
    }

/**
     * Retrieves the JPEG orientation from the specified screen rotation.
     *
     * @param rotation The screen rotation.
     * @return The JPEG orientation (one of 0, 90, 270, and 360)
     */
    private int getOrientation(int rotation,int mSensorOrientation) {
        // Sensor orientation is 90 for most devices, or 270 for some devices (eg. Nexus 5X)
        // We have to take that into account and rotate JPEG properly.
        // For devices with orientation of 90, we simply return our mapping from ORIENTATIONS.
        // For devices with orientation of 270, we need to rotate the JPEG 180 degrees.
        return (ORIENTATIONS.get(rotation) + mSensorOrientation + 270) % 360;
    }
`

change the line in **_Camera2_**

`captureRequestBuilder.set(CaptureRequest.JPEG_ORIENTATION,
                    (sensorOrientation +
                            mDisplayOrientation * (mFacing == Constants.FACING_FRONT ? 1 : -1) +
                            360) % 360);`

to this

`captureRequestBuilder.set(CaptureRequest.JPEG_ORIENTATION,
                    getOrientation(0,sensorOrientation));`

let me know if it works or not coz i dint have the above devices 

for the case of front camera if it is **_Camera1_**

` if (rotate == 0)
                        matrix.postRotate(180);
                    else if (rotate == 90)
                        matrix.postRotate(270);`

snowpong, 2017-02-21:
@yaraki I have the same problem on a Sony D6503 (Xperia Z2) running Android 6.0.1 (API23). The captured image is rotated 180 degrees.

@xp-vit Why shouldn't the library ensure the captured image has the right orientation? In my opinion any phone manufacturers idiosyncrasies should be ironed out by this library.

xp-vit, 2017-02-21:
@snowpong 

One of the reasons why I think that lib should NOT be rotating images itself, but just provide proper info is that Image rotation is memory consuming process and may fail with OOM anytime (we've seen this a lot in our app).That's why some developers may just decide to have this work done using their back-end services. However, in case if this is important, lib might need to have this behavior configurable.

BTW, have you tried solution provided by @jinujayakumar on that Sony D6503 (Xperia Z2)? I'm going to test this on Nexus 5X later today and will post the results.

snowpong, 2017-02-21:
@xp-vit OK, nobody wants OOM. But at the moment when we (the users of the library) get to the data it's a JPEG byte[] array, and there is no way we can rotate it without first reading it into a Bitmap and then creating another rotated one from it, which is sure to use a lot of memory. Ideally it'd be fixed earlier in the pipeline by the library if possible. 

I haven't tried @jinujayakumar suggestion yet. I think I'm just going to read the EXIF information (using the support library https://android-developers.googleblog.com/2016/12/introducing-the-exifinterface-support-library.html) and rotate myself.

xp-vit, 2017-02-21:
@snowpong Reading Exif just won't help. We have Exif reading in place, but for example on Nexus 5x Exif rotation = 0 when using Camera2, however image is upside down.

snowpong, 2017-02-21:
@xp-vit You're right. EXIF seems to be the same on the Z2 for both front and back facing, yet front facing is upside down.

mkotyk, 2017-02-24:
I'm seeing front facing camera orientation issues on Moto G and Samsung S5.  The Exif data does not have an orientation specified.

mkotyk, 2017-02-24:
@snowpong I found a cool library called LLJTran which will rotate a JPEG image without decompressing it.

snowpong, 2017-02-24:
@mkotyk Did you get it working?

I'm thinking @jinujayakumar might be on to something and that  https://github.com/googlesamples/android-Camera2Basic/blob/master/Application/src/main/java/com/example/android/camera2basic/Camera2BasicFragment.java#L829 should be used at https://github.com/google/cameraview/blob/master/library/src/main/api21/com/google/android/cameraview/Camera2.java#L608) but the Camera2Basic example explicitly does not support front facing cameras https://github.com/googlesamples/android-Camera2Basic/blob/master/Application/src/main/java/com/example/android/camera2basic/Camera2BasicFragment.java#L499 so I'm not sure if it's a proper solution for front facing - I haven't had time to test yet

mkotyk, 2017-02-24:
I'm seeing a possible issue in the Camera1 api.  The docs use two different formulas for calculating rotation.  One for `setDisplayOrientation` and one for `setRotation`.  In cameraview it uses the same function to calculate both.  Going to experiment with that a bit.  Haven't looked at what Camera2 does yet.

https://developer.android.com/reference/android/hardware/Camera.Parameters.html#setRotation(int)
https://developer.android.com/reference/android/hardware/Camera.html#setDisplayOrientation(int)

mkotyk, 2017-02-24:
Hmm I solved it for my situation on `Camera1`.  When switching to back/facing camera it wouldn't call `setRotation` because `if (currentSize.width != previewSize.getWidth() || currentSize.height != previewSize.getHeight()) {` would jump over that block.  That why I wasn't getting any orientation tag in the exif data.  I'll probably have a PR later today.

mkotyk, 2017-02-27:
I submitted a PR that fixed `Camera1` front and back orientation issues for all my test devices, which is sadly only 3.   I would appreciate any willing testers and feedback.

https://github.com/google/cameraview/pull/115

snowpong, 2017-02-28:
I now tested the latest commit (a9d1eb9bad93dd8da981d380e37f58e35f847af5) to capture a front facing photo on a Huawei Mate 9 Pro (API-24) and it's rotated by 90 degrees CW.

I haven't tested if @mkotyk PR fixes things as that is for Camera1 and my two devices use Camera2.

snowpong, 2017-03-08:
@mkotyk Just verified your PR fixes the 180 issue I had on front facing photos taken on a Sony Xperia Z2 (turns out it falls back to Camera1)

However, regarding the front facing issues I experience on my  Huawei Mate 9 Pro (API-24) , I've now tried all sorts of calculations to find the right JPEG_ORIENTATION to set in Camera2 (https://github.com/google/cameraview/blob/master/library/src/main/api21/com/google/android/cameraview/Camera2.java#L608) but it seems it doesn't matter what I specify there. The end result is the same.

I explicitly tried to set 0 and 90 directly, and verified the CaptureRequest contains this value for JPEG_ORIENTATION, but when I save the picture to file, the EXIF TAG_ORIENTATION stays the same and the displayed image has not changed in rotation.

It actually seems that whatever I put as JPEG_ORIENTATION is ignored?

I've verified by debugging and Log output that my codepath is actually run. Any ideas or similar experiences? Shouldn't changing the JPEG_ORIENTATION either result in an EXIF TAG_ORIENTATION change, or an actual change in the JPEG file saved? @xp-vit and @jinujayakumar did any of you resolve your Camera2 front facing issues?

snowpong, 2017-03-09:
I moved on to https://github.com/flurgle/CameraKit-Android (a fork of this project). Front facing camera works fine there.