google/cameraview #85: Preview not display in Nougat

ShreyashPromact, 2017-01-23:
I have Nexus 6P with Nougat OS in it. I have run the demo in this device. But it is not displaying a preview. I have checked the same demo in other OS and devices but it is working there.

So, I guess the issue is with either Nougat OS support or with the device. But device camera within other app is also working fine.
So, finally, the issue is with support for Nougat OS in Nexus 6P.
Can you please check that and let me know when the issue gets resolved.

Sincerely,
Shreyash

rajeshct, 2017-01-25:
Go to application info did it granted permission for Camera and storage.

ShreyashPromact, 2017-01-25:
@rajeshct Yes, it is already granted. Can you please try that same and let me know.

I have pre-confirm that the permission is already granted and approved by the user. It is not opening the camera feed and when I click capture button, it crashed the app.

Sincerely,
Shreyash

rajeshct, 2017-01-26:
Check Front Camera issue your issue resolved

ShreyashPromact, 2017-01-27:
@rajeshct 
Thanks for the reply but the issue is not like that.

Your issue is with the front camera. As you have mentioned there, 

> "Image rotated 180 degrees while capturing from front camera"

I have read all the comments on that issue. But here the problem is different. Here Camera is not displayed. The whole screen looks like black. When I capture the image with this behavior, it crashes the app.

Sincerely,
Shreyash

rajeshct, 2017-01-27:
Check [](https://github.com/rajeshct/cameraview)

rajeshct, 2017-01-27:
Check the link if it resolved error let me know
[Click Here](https://github.com/rajeshct/cameraview)

ShreyashPromact, 2017-01-27:
@rajeshct 
I have updated the library with latest one and now it is working fine.

The issue seems like with the aspect-ratio. I can able to see the camera but it is not in full screen. I want it to have full screen.

Can you please give me the suggestion for how to make it full screen?

Please check attached screen-shot of my app for camera screen I got in Nougat.

http://imgur.com/a/DbljT

Sincerely,
Shryash

ShreyashPromact, 2017-01-27:
@rajeshct 

I have seen that the library now have aspect ratio to display camera preview. Default aspect ratio is 4:3 as mention in Constants.java file of the library.

I would like to have the preview in full screen for all device. How can I achieve that?

Before this, I have to make layout height of Camera to match_parent which works fine. But that is now not working in Nougat device. I need to make it wrap_content and use aspect ratio to display camera preview.

Please provide me the possible workout to display camera preview full screen. I have tried with another aspect ratio (like.. 5:3, 9:5, etc) but it seems like not working properly.

Sincerely,
Shreyash

ShreyashPromact, 2017-01-30:
@rajeshct 

Hey, Do you have solution related this? If not then I can raise another issue here in this library with the new thread. Now, the issue is related to full-screen preview.

Sincerely,
Shreyash

rajeshct, 2017-01-30:
Sorry but didn't get your point aspect ratio just related to captured image it doesn't deal with preview. Preview totally depend upon how you want to display it. Do this in your layout 
```
<FrameLayout
    xmlns:android="http://schemas.android.com/apk/res/android"
    xmlns:tools="http://schemas.android.com/tools"
    android:layout_width="match_parent"
    android:layout_height="match_parent"
    android:orientation="vertical"
    tools:context=".MainActivity">

    <include
        layout="@layout/include_camera"
        android:layout_width="match_parent"
        android:layout_height="match_parent"/>

    <include
        layout="@layout/include_control"
        android:layout_width="wrap_content"
        android:layout_height="wrap_content"
        android:layout_gravity="bottom|center"/>

</FrameLayout>
```

ShreyashPromact, 2017-01-31:
@rajeshct 

My Code was like that before and it was working fine. But that was not working in Nougat. After the update of Library, it is working in Nougat, but the issue is, I need to keep it in the aspect ratio given by library it self. If I remove aspect ration, it stop working. I am not able to see the preview.

Now, Whether I have to make CameraView as match_parent or as `wrap_content` ? If I keep it as `wrap_contect `then How I can able to see the preview in fullScreen.

Sincerely,
Shreyash

ataulm, 2017-02-02:
It would make sense to close this and open a new issue if the problem is something different.

ShreyashPromact, 2017-02-03:
@ataulm @rajeshct 
#98 
I have created the new thread for this issue. Please participate to get me out from this issue.
Thanks.

closed 2017-02-03 by ShreyashPromact