google/cameraview #86: CameraView takes time to capture image

protocol10, 2017-01-25:
- Camera View takes time to capture the image if we request `cameraview.takePicture()` it takes about 1 or 3 secs to receive callback for onPictureTaken Callback 

Found on Asus Zenphone

joshgosse, 2017-06-27:
I've been trying to figure out what has been going on here for a couple of weeks (albeit I haven't spent a lot of time), glad to see it might just be an issue with Asus Zenfone!

3dm1, 2017-07-26:
I was having the same issue with a [HTC One (M8)](http://www.gsmarena.com/htc_one_(m8)-6074.php) and it turned out to be caused by setting auto-focus to true.

joshgosse, 2017-07-26:
It turns out that in the end the camera on Android is just pretty slow overall. That being said, I ended up switching to Fotoapparat: https://github.com/Fotoapparat/Fotoapparat and have found there to be less crashes on various devices