google/cameraview #90: on resume than get this exception

deepikaopenxcell, 2017-01-27:
java.lang.SecurityException: Lacking privileges to access camera service
                                                                  at android.hardware.camera2.utils.CameraBinderDecorator.throwOnError(CameraBinderDecorator.java:108)
                                                                  at android.hardware.camera2.utils.CameraBinderDecorator$CameraBinderDecoratorListener.onAfterInvocation(CameraBinderDecorator.java:73)
                                                                  at android.hardware.camera2.utils.Decorator.invoke(Decorator.java:81)
                                                                  at java.lang.reflect.Proxy.invoke(Proxy.java:393)
                                                                  at $Proxy5.createDefaultRequest(Unknown Source)
                                                                  at android.hardware.camera2.impl.CameraDeviceImpl.createCaptureRequest(CameraDeviceImpl.java:628)
                                                                  at com.infield.customcamera.Camera2.startCaptureSession(Camera2.java:603)
                                                                  at com.infield.customcamera.Camera2$1.onOpened(Camera2.java:136)
                                                                  at android.hardware.camera2.impl.CameraDeviceImpl$1.run(CameraDeviceImpl.java:134)
                                                                  at android.os.Handler.handleCallback(Handler.java:739)
                                                                  at android.os.Handler.dispatchMessage(Handler.java:95)
                                                                  at android.os.Looper.loop(Looper.java:148)
                                                                  at android.app.ActivityThread.main(ActivityThread.java:5417)
                                                                  at java.lang.reflect.Method.invoke(Native Method)
                                                                  at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:726)
                                                                  at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:616)

jaksab, 2017-01-28:
User must accept the android camera permission before you try to open the camera

jf-rce, 2017-01-30:
This is a duplicate of https://github.com/google/cameraview/issues/61

It happens even when the camera permission has been granted. It seems to occur, for me, when a Dialog is shown above then CameraView and then removed.

This issue exists in the Google Camera2Basic example as well. My recommendations in the issue #61 thread fixed the issue on some offending devices. However, I have still yet to fix the problem on the Galaxy S6. I am looking into using the Samsung Camera API to potentially mitigate the problem.

deepikaopenxcell, 2017-01-31:
@jf-rce  yes I am using Samsung device which I tested s3 and s4 and s7 there are so much bug. I got one more bug Take picture failed also.