google/cameraview #96: takePicture failed while taking picture on the move(Moto G3)

jydimir, 2017-02-02:
FATAL EXCEPTION: main
    Process: com.masterofcode.llc.phoxapp.debug, PID: 7007
    java.lang.RuntimeException: takePicture failed
    at android.hardware.Camera.native_takePicture(Native Method)
    at android.hardware.Camera.takePicture(Camera.java:1461)
    at com.google.android.cameraview.Camera1.takePictureInternal(Camera1.java:230)
    at com.google.android.cameraview.Camera1$2.onAutoFocus(Camera1.java:221)
    at android.hardware.Camera$EventHandler.handleMessage(Camera.java:1146)
    at android.os.Handler.dispatchMessage(Handler.java:102)
    at android.os.Looper.loop(Looper.java:148)
    at android.app.ActivityThread.main(ActivityThread.java:5443)
    at java.lang.reflect.Method.invoke(Native Method)
    at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:728)
    at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:618)

kevinpelgrims, 2017-02-08:
I'm seeing the same issue in crash reports. Reported devices are (among others) Sony C2304 (Android 4.2.2), Huawei KII-L21 (Android 6.0.1), and Oppo A33w (Android 5.1).

This is the full stacktrace:
```
Fatal Exception: java.lang.RuntimeException: takePicture failed
       at android.hardware.Camera.native_takePicture(Camera.java)
       at android.hardware.Camera.takePicture(Camera.java:1647)
       at com.google.android.cameraview.Camera1.takePictureInternal(Camera1.java:230)
       at com.google.android.cameraview.Camera1$2.onAutoFocus(Camera1.java:221)
       at android.hardware.Camera$EventHandler.handleMessage(Camera.java:1216)
       at android.os.Handler.dispatchMessage(Handler.java:102)
       at android.os.Looper.loop(Looper.java:150)
       at android.app.ActivityThread.main(ActivityThread.java:5546)
       at java.lang.reflect.Method.invoke(Method.java)
       at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:794)
       at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:684)
```

Some of the crashes include this: `Fatal Exception: java.lang.RuntimeException: takePicture failed, error=-38`

kevinpelgrims, 2017-02-08:
@jydimir can you reproduce this consistently? If so, can you explain how.

jydimir, 2017-02-11:
@kevinpelgrims It was reproduceable. I think problems with a focus on Camera 1: `at com.google.android.cameraview.Camera1$2.onAutoFocus(Camera1.java:221) `
When I took picture while I was moving camera quickly I got that exception.
I needed a quick fix, so I reverted https://github.com/google/cameraview/commit/f0eef9ade1969f9dafef7138172468e409c3210c because fell back to Camera 1 even when api was >= 21.

Debdeep009, 2017-12-20:
error code -38 resembles to a constant `ENOSYS` in `OsSystem` class present in android.system package. `ENOSYS` is thrown when some of the camera methods are not implemented.