google/cameraview #98: Camera preview not working as fullscreen in Nexus 6P with Nougat OS

ShreyashPromact, 2017-02-03:
I have raised this issue here in this thread: #85 
But as it is suggested to start new thread, I have created this one.

@ataulm @rajeshct
I want you to participate in this and help me to resolved this issue.

The issue is that, I am not able to see the preview in Nexus 6P with Nougat OS (Here I want to see the preview in full screen only). 

Earlier I have made my camera layout as follow to make it full screen:
```
<include
        layout="@layout/include_camera"
        android:layout_width="match_parent"
        android:layout_height="match_parent"/>
```

With this, it was working fine in all the device except Nexus 6p with Nougat OS. After that, I have updated the library and check for the issue. Here the Issue is get resolved but I need to make camera layout as per below code and also need to add AspectRatio to the Camera preview to display in that ratio.
```
<include
        layout="@layout/include_camera"
        android:layout_width="match_parent"
        android:layout_height="wrap_content"/>
```

Now, If I make it to `match_parent` in height then It will stop showing preview in any device (Not only 6P with Nougat OS). And If I make it `wrap_content` then, it will show preview, but not in full screen but with the aspect ration of the screen.

So, here I want to have the full-screen preview of the Camera which was working nicely in old devices (Lower then 6p and Nougat OS).

I hope you understand the issue I am facing with the update of the library.

Please help me to resolved this issue.

ShreyashPromact, 2017-02-08:
@rajeshct @ataulm

I need any input or solution for this queries.
Please need help.

Sincerely,
Shreyash

ataulm, 2017-02-08:
@ShreyashPromact I am confused by the following

>I need to make camera layout as per below code and also need to add AspectRatio to the Camera preview to display in that ratio.
>
>```xml
><include
>    layout="@layout/include_camera"
>    android:layout_width="match_parent"
>    android:layout_height="wrap_content" />
>```
>...
>So, here I want to have the full-screen preview of the Camera 

Do you want it full screen or wrap content to conform to a specific aspect ratio?

ShreyashPromact, 2017-02-08:
@ataulm

I want to display it in full screen. I know `match_parent` is the key solution for that. But If I keep it like that, Camera View is not working in Nexus 6P device with Nougat OS while same code working in Motorola's device.

Sincerely,
Shreyash

ataulm, 2017-02-08:
I don't have a 6p so I can't debug this issue, sorry.

ShreyashPromact, 2017-02-20:
Can anyone help me to resolved this issue?

This issue is only in Nexus6p device with Nougat OS.

Sincerely,
Shreyash

rajeshct, 2017-02-20:
Ok I will check this in emulator because I didn't have nexus

ShreyashPromact, 2017-02-28:
@rajeshct 

Thanks a lot.

ShreyashPromact, 2017-02-28:
@rajeshct 
It might not possible to check in emulator, especially for Camera. Can you please check it in real device to get from any other and update this.
Thanks.

Sincerely,
Shreyash

ShreyashPromact, 2017-03-14:
@rajeshct 

I am getting lots of update regarding this project. but not getting any updates for this raised issue. Please can anyone look into this issue and update it. Please assign this to any developer to check it and resolved it.

Sincerely,
Shreyash

ShreyashPromact, 2017-04-06:
The issue is resolved.

closed 2017-04-06 by ShreyashPromact