google/cameraview #111: Cameraview blacks out after changing its size

ankitbatra11, 2017-02-20:
Hi,

cameraview blacks out after changing it's LayoutParams in an activity. Below piece of code used to work earlier before I updated to the latest changes:

`public void applySize(int sizePixels) {
        if (cameraViewRef != null) {
            CameraView cameraView = cameraViewRef.get();
            if (cameraView != null) {
                ViewGroup.LayoutParams params = cameraView.getLayoutParams();
                if (params != null) {
                    params.width = sizePixels;
                    params.height = sizePixels;
                    cameraView.setLayoutParams(params);
                } else {
                    Timber.d("layout params are null for cameraview");
                }
            } else {
                Timber.d("camera view got GC'ed");
            }
        } else {
            Timber.e("must set a camera view");
        }
    }`

mkotyk, 2017-03-07:
I see that also when changing `Visibility`

FabinPaul, 2017-03-24:
@mkotyk `TextureView` is added twice to `CameraView`. So when `requestLayout` is called on `CameraView` blank surface comes up. Note that `CameraView` parent is `FrameLayout`

csbz17027, 2017-08-16:
I met the same requirements and issues.
You can replace the layout to achieve the effect.