google/cameraview #146: Error when resuming activity

yperess, 2017-04-26:
As mentioned in the README, I start the camera in onResume (conditioned on a shared preferences flag and camera permissions) and stop the camera (if it was started) in onPause. This is crashing when using activities with singleTop set. When I get a call to onNewIntent the activity is resumed, paused, then resumed again. This causes the following error:

```
W/System.err: java.lang.RuntimeException: Failed to start camera session
W/System.err:     at com.google.android.cameraview.Camera2.startCaptureSession(Camera2.java:473)
W/System.err:     at com.google.android.cameraview.Camera2$1.onOpened(Camera2.java:75)
W/System.err:     at android.hardware.camera2.impl.CameraDeviceImpl$1.run(CameraDeviceImpl.java:134)
W/System.err:     at android.os.Handler.handleCallback(Handler.java:751)
W/System.err:     at android.os.Handler.dispatchMessage(Handler.java:95)
W/System.err:     at android.os.Looper.loop(Looper.java:154)
W/System.err:     at android.app.ActivityThread.main(ActivityThread.java:6121)
W/System.err:     at java.lang.reflect.Method.invoke(Native Method)
W/System.err:     at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:889)
W/System.err:     at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:779)
```

I've also tried making use of the CameraView.Callback to queue up operations in case the toggle caused by pause/resume was messing up the camera state. This didn't seem to do anything.

yperess, 2017-04-26:
Update, I was able to replicate this by simply posting mCameraView.start() to the Main thread with 250 ms delay in onResume (no need for onNewIntent)

yperess, 2017-04-28:
This appears to be related to calling `start()`, `stop()`, then `start()` again. I have a potential fix, and will post a PR this weekend.

saulmm, 2017-05-25:
@yperess did you sent the PR?, I'm interested on the fix. Experimenting the same after coming from a system camera activity and firing `cameraView.start()`

yperess, 2017-05-25:
@saulmm Sorry, I got caught up with deadlines and the fix ended up being more complex than I expected. I'll do my best to contain it and post it in the next few days.

yperess, 2017-05-25:
@saulmm Wow, I think that taking a look again I might have found the actual issue and my complex workaround was useless. Let me know if that PR works for you (seems to work for me).

saulmm, 2017-05-26:
Mmm, unfortunately @yperess, it doesn't work for me. Looks like `Camera2` is throwing a `CameraException` with the 'disconnected' message when coming from an intent chooser which has fired the system camera. [Here](https://github.com/google/cameraview/blob/master/library/src/main/api21/com/google/android/cameraview/Camera2.java#L486).

It seems that the system camera needs some delay for properly access to another instance again, since workarounding in this way works:

```
@Override
    public void onResume() {
        super.onResume();

        if (!mHasFailImageRequestResult) {
            mBinding.camera.start();

        } else { // Workaround for the CameraException thrown by CameraView with a disconnected camera message, https://goo.gl/ymgeJp
            new Handler().postDelayed(() -> {
                if (getActivity() != null && !mBinding.camera.isCameraOpened())
                    mBinding.camera.start();
            }, DELAY_COMING_FROM_DEVICE_CAMERA);
        }
    }
```

This is an awful hack so I'll keep trying finding how to work around these issue in an elegant way. Thank you for pinging me your solution anyway 👍.

Maybe @yaraki could put some light on to this :)

yperess, 2017-05-26:
I'll post my old solution for you tomorrow when I have a bit. Maybe it's
more appropriate for you. I basically registered a listener with the
CameraView and queued up operations (squashing states that aren't needed).

Basically if you call start() then stop(), then stop won't get called until
the callback was called saying the camera was started. On errors I just pop
the stop. If I get another stop (maybe due to lifecycle change I drop it),
and if I get a start I pop the stop and drop the start.