google/cameraview #147: Bump minSdkVersion to 14

yaraki, 2017-04-27:
Drop support for Camera1+SurfaceView combination.

We might be able to use ActivityLifecycleCallbacks to monitor callbacks. Users don't need to call to call CameraView#onStart and onStop
https://developer.android.com/reference/android/app/Application.html#registerActivityLifecycleCallbacks(android.app.Application.ActivityLifecycleCallbacks)

bemusementpark, 2017-04-28:
For a complex app where `CameraView`s may be added and removed I can see two ways to make this work, yet both involve some drawbacks.

1. incorporate the attached state of the `CameraView` into 
call `onStart` when the `CameraView` has been attached **AND** the Activity is resumed.
call `onStop` when `CameraView` has been removed **OR** the Activity is paused.

cons: this would mean that if a `CameraView` was detached and promptly re-attached it would have to undergo the costly `CameraView.onStart()` method, even if the developer could currently avoid this.

2. Ignore the attached state of the `CameraView`.
cons: 
i. This would consume resources, battery and the camera for as long as the `Activity` remains resumed.
ii. No other `CameraView` could use the camera being in use by the previously removed `CameraView`.

In either implementation, the drawbacks are sufficiently burdensome for me to encourage you to drop this feature or make it optional.

tikimcfee, 2017-05-03:
Hey there! Just adding my two-cents. I agree with @bemusementpark . A lot of devs (myself, team, and dev community included!) are very much interested in a _stable, basic, and wrappable CameraView implementation_. 

There is already an understood expectation around libs like this to have to tunnel lifecycle events down to where they need to be (Google's MapView, for example), so it's not a big deal for most of the community. The stuff that _does_ make a lot of sense are the very careful decisions about when to catch exceptions, when to callbacks to errors, when to retry things like focus, etc.