google/cameraview #152: Aspect Ratio option not working

anantprsd5, 2017-05-12:
When trying to change the aspect ratio, the camera view doesn't work and a black screen comes in its place

dk1206, 2017-05-12:
Hey Anantprsd5,

Are you setting one of the available Aspect Ratios for your mobile phone? Could you maybe also share some code with your issue.

anantprsd5, 2017-05-12:
@dk1206 Yes. I am using the demo application of the camera view and when I click on the menu to change the aspect ratio, I am getting a black screen like the screenshot below.
[image]

dk1206, 2017-05-12:
I cannot reproduce it with my phone, for me it works. But that is rather curious aspect ratio you have btw.

If you can edit your initial message, please add the also the Mobile phone model you are using and also write there the OS version.

anantprsd5, 2017-05-12:
@dk1206 Done.

aalices, 2017-06-29:
The same issue for Xiaomi Redmi 4A, Android: 6.0.1