google/cameraview #154: Autofocus is broken on Galaxy S7

jryburn2712, 2017-05-13:
Auto focus randomly doesn't work on the S7's back-facing camera (I haven't tested the front). I've been looking for a pattern but I can't find any, except that it seems like pausing and resuming the activity is a near guarantee that auto focus won't work (calling cameraView.stop() in onPause() and starting it in onResume()). Starting the activity for the first time is 50/50. Sometimes auto focus works, and sometimes it doesn't. I've tried setting the auto focus in both xml and java (including both at the same time) and it doesn't make a difference. I've tested the exact code on an older phone (Droid RAZR, 19) and it works fine.

cwliu, 2017-05-19:
I also notified this issue reported from Firebase crash reporting. It's also on Samsung Galaxy S7 model: 

```
Exception java.lang.NullPointerException: Attempt to invoke virtual method 'int android.hardware.camera2.CameraCaptureSession.capture(android.hardware.camera2.CaptureRequest, android.hardware.camera2.CameraCaptureSession$CaptureCallback, android.os.Handler)' on a null object reference
com.google.android.cameraview.Camera2.unlockFocus (Camera2.java:661)
com.google.android.cameraview.Camera2$6.onCaptureCompleted (Camera2.java:645)
java.lang.reflect.Method.invoke (Method.java)
android.hardware.camera2.dispatch.InvokeDispatcher.dispatch (InvokeDispatcher.java:39)
android.hardware.camera2.dispatch.HandlerDispatcher$1.run (HandlerDispatcher.java:65)
android.os.Handler.handleCallback (Handler.java:751)
android.os.Handler.dispatchMessage (Handler.java:95)
android.os.Looper.loop (Looper.java:154)
android.app.ActivityThread.main (ActivityThread.java:6688)
java.lang.reflect.Method.invoke (Method.java)
com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run (ZygoteInit.java:1468)
com.android.internal.os.ZygoteInit.main (ZygoteInit.java:1358)

```
[image]

dilix, 2017-06-30:
Provided stack trace happen because session is not initialized yet but you try to 'take picture'.
It's easy to reproduce on many devices by pressing 'rotate camera' + 'take picture' at the same time.

This fix helps me:
#175

RohanVDeshpande, 2018-06-13:
Has anyone found a solution to this issue? I am experiencing the same issues with my Galaxy S7...