google/cameraview #156: Camera is not ready. Call start() before takePicture.

vijaysshelke, 2017-05-14:
Faced new issue Camera is not ready. Call start() before takePicture. Please provide solution. This issue came in camera1 API.

dk1206, 2017-05-15:
Hello vijazsshelke!

What do you mean, what kind of issue is exactly occuring to you?

You have to call camera.start() before taking pictures, that holds true, yes. You have to do it in onResume() lifecycle method of Android. 

That is how it works.

aswalalka, 2017-07-05:
I am also facing the same issue, it comes randomly, how to fix it?

xiaanming, 2017-08-14:
Click on the photo button quickly can load to crash

xiaanming, 2017-08-14:
To start an new preview activity before take pictrue success, then click on the photo button quickly can load to crash