google/cameraview #161: Failed to start camera session when switching to front facing camera

ejdelros08, 2017-05-30:
Camera just hangs when switching from Back facing camera to Front facing camera on Xperia Z3+ (E6553) Android version 6.0.1

Error Trace:

java.lang.RuntimeException: Failed to start camera session
	at com.google.android.cameraview.Camera2.startCaptureSession(Camera2.java:486)
	at com.google.android.cameraview.Camera2$5.onSurfaceChanged(Camera2.java:205)
	at com.google.android.cameraview.PreviewImpl.dispatchSurfaceChanged(PreviewImpl.java:54)
	at com.google.android.cameraview.TextureViewPreview$1.onSurfaceTextureSizeChanged(TextureViewPreview.java:51)
	at android.view.TextureView.onSizeChanged(TextureView.java:329)
	at android.view.View.sizeChange(View.java:16840)
	at android.view.View.setFrame(View.java:16802)
	at android.view.View.layout(View.java:16718)
	at android.widget.FrameLayout.layoutChildren(FrameLayout.java:336)
	at android.widget.FrameLayout.onLayout(FrameLayout.java:273)
	at android.view.View.layout(View.java:16722)
	at android.view.ViewGroup.layout(ViewGroup.java:5438)
	at android.widget.RelativeLayout.onLayout(RelativeLayout.java:1080)
	at android.view.View.layout(View.java:16722)
	at android.view.ViewGroup.layout(ViewGroup.java:5438)
	at android.widget.FrameLayout.layoutChildren(FrameLayout.java:336)
	at android.widget.FrameLayout.onLayout(FrameLayout.java:273)
	at android.view.View.layout(View.java:16722)
	at android.view.ViewGroup.layout(ViewGroup.java:5438)
	at android.support.v7.widget.ActionBarOverlayLayout.onLayout(ActionBarOverlayLayout.java:435)
	at android.view.View.layout(View.java:16722)
	at android.view.ViewGroup.layout(ViewGroup.java:5438)
	at android.widget.FrameLayout.layoutChildren(FrameLayout.java:336)
	at android.widget.FrameLayout.onLayout(FrameLayout.java:273)
	at android.view.View.layout(View.java:16722)
	at android.view.ViewGroup.layout(ViewGroup.java:5438)
	at android.widget.LinearLayout.setChildFrame(LinearLayout.java:1743)
	at android.widget.LinearLayout.layoutVertical(LinearLayout.java:1586)
	at android.widget.LinearLayout.onLayout(LinearLayout.java:1495)
	at android.view.View.layout(View.java:16722)
	at android.view.ViewGroup.layout(ViewGroup.java:5438)
	at android.widget.FrameLayout.layoutChildren(FrameLayout.java:336)
	at android.widget.FrameLayout.onLayout(FrameLayout.java:273)
	at com.android.internal.policy.PhoneWindow$DecorView.onLayout(PhoneWindow.java:2685)
	at android.view.View.layout(View.java:16722)
	at android.view.ViewGroup.layout(ViewGroup.java:5438)
	at android.view.ViewRootImpl.performLayout(ViewRootImpl.java:2175)
	at android.view.ViewRootImpl.performTraversals(ViewRootImpl.java:1935)
	at android.view.ViewRootImpl.doTraversal(ViewRootImpl.java:1111)
	at android.view.ViewRootImpl$TraversalRunnable.run(ViewRootImpl.java:6017)
	at android.view.Choreographer$CallbackRecord.run(Choreographer.java:858)
	at android.view.Choreographer.doCallbacks(Choreographer.java:670)
	at android.view.Choreographer.doFrame(Choreographer.java:606)
	at android.view.Choreographer$FrameDisplayEventReceiver.run(Choreographer.java:844)
	at android.os.Handler.handleCallback(Handler.java:739)
	at android.os.Handler.dispatchMessage(Handler.java:95)
	at android.os.Looper.loop(Looper.java:234)
	at android.app.ActivityThread.main(ActivityThread.java:5526)
	at java.lang.reflect.Method.invoke(Native Method)
	at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:726)
	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:616)

TheMelody, 2017-09-04:
Build. Version. SDK_INT > = 21, need through the CameraManager openCamera(xx,xx,xx) method sets the front-facing camera and rear camera, cameraId for String type at the same time, lead to "1", set to "0" after.
**## mCameraManager.openCamera(cameraId,mCameraDeviceCallback,null);**

ldemyanenko, 2018-01-25:
Have the same issue. Is there any solution for this?
Looks like this issue appears only on Android 7.

Fatal Exception: java.lang.RuntimeException: Failed to start camera session
       at com.google.android.cameraview.Camera2.startCaptureSession(Camera2.java:491)
       at com.google.android.cameraview.Camera2$5.onSurfaceChanged(Camera2.java:210)
       at com.google.android.cameraview.PreviewImpl.dispatchSurfaceChanged(PreviewImpl.java:54)
       at com.google.android.cameraview.TextureViewPreview$1.onSurfaceTextureSizeChanged(TextureViewPreview.java:51)
       at android.view.TextureView.onSizeChanged(TextureView.java:366)
       at android.view.View.sizeChange(View.java:18941)
       at android.view.View.setFrame(View.java:18884)
       at android.view.View.layout(View.java:18797)
       at android.widget.FrameLayout.layoutChildren(FrameLayout.java:323)
       at android.widget.FrameLayout.onLayout(FrameLayout.java:261)
       at android.view.View.layout(View.java:18804)
       at android.view.ViewGroup.layout(ViewGroup.java:5951)
       at android.support.constraint.ConstraintLayout.onLayout(ConstraintLayout.java:1197)
       at android.view.View.layout(View.java:18804)
       at android.view.ViewGroup.layout(ViewGroup.java:5951)
       at android.widget.FrameLayout.layoutChildren(FrameLayout.java:323)
       at android.widget.FrameLayout.onLayout(FrameLayout.java:261)
       at android.view.View.layout(View.java:18804)
       at android.view.ViewGroup.layout(ViewGroup.java:5951)
       at android.widget.LinearLayout.setChildFrame(LinearLayout.java:1741)
       at android.widget.LinearLayout.layoutVertical(LinearLayout.java:1585)
       at android.widget.LinearLayout.onLayout(LinearLayout.java:1494)
       at android.view.View.layout(View.java:18804)
       at android.view.ViewGroup.layout(ViewGroup.java:5951)
       at android.widget.FrameLayout.layoutChildren(FrameLayout.java:323)
       at android.widget.FrameLayout.onLayout(FrameLayout.java:261)
       at android.view.View.layout(View.java:18804)
       at android.view.ViewGroup.layout(ViewGroup.java:5951)
       at android.widget.LinearLayout.setChildFrame(LinearLayout.java:1741)
       at android.widget.LinearLayout.layoutVertical(LinearLayout.java:1585)
       at android.widget.LinearLayout.onLayout(LinearLayout.java:1494)
       at android.view.View.layout(View.java:18804)
       at android.view.ViewGroup.layout(ViewGroup.java:5951)
       at android.widget.FrameLayout.layoutChildren(FrameLayout.java:323)
       at android.widget.FrameLayout.onLayout(FrameLayout.java:261)
       at com.android.internal.policy.DecorView.onLayout(DecorView.java:891)
       at android.view.View.layout(View.java:18804)
       at android.view.ViewGroup.layout(ViewGroup.java:5951)
       at android.view.ViewRootImpl.performLayout(ViewRootImpl.java:2659)
       at android.view.ViewRootImpl.performTraversals(ViewRootImpl.java:2375)
       at android.view.ViewRootImpl.doTraversal(ViewRootImpl.java:1526)
       at android.view.ViewRootImpl$TraversalRunnable.run(ViewRootImpl.java:7102)
       at android.view.Choreographer$CallbackRecord.run(Choreographer.java:927)
       at android.view.Choreographer.doCallbacks(Choreographer.java:702)
       at android.view.Choreographer.doFrame(Choreographer.java:638)
       at android.view.Choreographer$FrameDisplayEventReceiver.run(Choreographer.java:913)
       at android.os.Handler.handleCallback(Handler.java:751)
       at android.os.Handler.dispatchMessage(Handler.java:95)
       at android.os.Looper.loop(Looper.java:154)
       at android.app.ActivityThread.main(ActivityThread.java:6682)
       at java.lang.reflect.Method.invoke(Method.java)
       at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:1520)
       at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:1410)