google/cameraview #170: Samsung Galaxy S6 freezes with flash always on

stonecompass, 2017-06-20:
When setting the flash to on the preview just freezes, when trying to take a picture. Is there a workaround for this?

stonecompass, 2017-06-21:
I can work around this when just setting it to use Camera API 1.

Ashky, 2017-06-23:
I have a Samsung Galaxy S6 and i using the camera2 Api, this the way i fixed it:
```
private void setAutoFlash(CaptureRequest.Builder requestBuilder) {
        if (mFlashSupported) {
            requestBuilder.set(CaptureRequest.CONTROL_AF_MODE, CaptureRequest.CONTROL_AE_MODE_OFF);
            requestBuilder.set(CaptureRequest.FLASH_MODE,CameraMetadata.FLASH_MODE_OFF);

            if(flashON)
                requestBuilder.set(CaptureRequest.FLASH_MODE,CameraMetadata.FLASH_MODE_TORCH);
        }
    }
```

GuillaumeLachaud, 2017-06-23:
I can reproduce on a Galaxy S6 and a Galaxy Note 5 (SM-N920V). Can repro in the sample app, 
setting the flash to auto works. When attempting to take a picture, logcat ouputs a lot of stuff in a loop:
```06-23 19:46:23.641  3166 17570 W ExynosCamera3: [CAM_ID(0)][]-WARN(m_previewStream3AAPipeThreadFunc):wait timeout
06-23 19:46:23.641  3166 17571 W ExynosCamera3: [CAM_ID(0)][]-WARN(m_previewStreamISPPipeThreadFunc):wait timeout
06-23 19:46:23.641  3166 17572 W ExynosCamera3: [CAM_ID(0)][]-WARN(m_duplicateBufferThreadFunc):wait timeout
06-23 19:46:23.791  3166 17570 W ExynosCamera3: [CAM_ID(0)][]-WARN(m_previewStream3AAPipeThreadFunc):wait timeout
06-23 19:46:23.791  3166 17571 W ExynosCamera3: [CAM_ID(0)][]-WARN(m_previewStreamISPPipeThreadFunc):wait timeout
06-23 19:46:23.791  3166 17572 W ExynosCamera3: [CAM_ID(0)][]-WARN(m_duplicateBufferThreadFunc):wait timeout
06-23 19:46:23.791  3586  4436 I EDMNativeHelperService: isCameraEnabled
06-23 19:46:23.941  3166 17571 W ExynosCamera3: [CAM_ID(0)][]-WARN(m_previewStreamISPPipeThreadFunc):wait timeout
06-23 19:46:23.941  3166 17572 W ExynosCamera3: [CAM_ID(0)][]-WARN(m_duplicateBufferThreadFunc):wait timeout
06-23 19:46:23.941  3166 17570 W ExynosCamera3: [CAM_ID(0)][]-WARN(m_previewStream3AAPipeThreadFunc):wait timeout
06-23 19:46:23.951  3166 17566 W ExynosCameraMCPipe: [CAM_ID(0)][PIPE_3AA]-WARN(m_putBuffer[1392]):inputFrameQ wait timeout
06-23 19:46:23.951  3166 17566 W ExynosCameraMCPipe: [CAM_ID(0)][PIPE_3AA]-WARN(m_putBufferThreadFunc[1348]):m_putbuffer fail, ret(-110)
06-23 19:46:24.001  3166 17568 W ExynosCameraMCPipe: [CAM_ID(0)][PIPE_ISP]-WARN(m_putBuffer[1392]):inputFrameQ wait timeout
06-23 19:46:24.001  3166 17568 W ExynosCameraMCPipe: [CAM_ID(0)][PIPE_ISP]-WARN(m_putBufferThreadFunc[1348]):m_putbuffer fail, ret(-110)
06-23 19:46:24.051  3166 17567 W ExynosCameraMCPipe: [CAM_ID(0)][PIPE_3AA]-WARN(m_getBuffer[1696]):requestFrameQ wait timeout
06-23 19:46:24.051  3166 17567 W ExynosCameraMCPipe: [CAM_ID(0)][PIPE_3AA]-WARN(m_getBufferThreadFunc[1364]):m_getBuffer fail, ret(-110)
06-23 19:46:24.051  3166 17569 W ExynosCameraMCPipe: [CAM_ID(0)][PIPE_ISP]-WARN(m_getBuffer[1696]):requestFrameQ wait timeout
06-23 19:46:24.051  3166 17569 W ExynosCameraMCPipe: [CAM_ID(0)][PIPE_ISP]-WARN(m_getBufferThreadFunc[1364]):m_getBuffer fail, ret(-110)
06-23 19:46:24.091  3166 17571 W ExynosCamera3: [CAM_ID(0)][]-WARN(m_previewStreamISPPipeThreadFunc):wait timeout
06-23 19:46:24.091  3166 17572 W ExynosCamera3: [CAM_ID(0)][]-WARN(m_duplicateBufferThreadFunc):wait timeout
06-23 19:46:24.091  3166 17570 W ExynosCamera3: [CAM_ID(0)][]-WARN(m_previewStream3AAPipeThreadFunc):wait timeout
06-23 19:46:24.241  3166 17571 W ExynosCamera3: [CAM_ID(0)][]-WARN(m_previewStreamISPPipeThreadFunc):wait timeout
06-23 19:46:24.241  3166 17570 W ExynosCamera3: [CAM_ID(0)][]-WARN(m_previewStream3AAPipeThreadFunc):wait timeout
06-23 19:46:24.241  3166 17572 W ExynosCamera3: [CAM_ID(0)][]-WARN(m_duplicateBufferThreadFunc):wait timeout
06-23 19:46:24.391  3166 17571 W ExynosCamera3: [CAM_ID(0)][]-WARN(m_previewStreamISPPipeThreadFunc):wait timeout
06-23 19:46:24.391  3166 17572 W ExynosCamera3: [CAM_ID(0)][]-WARN(m_duplicateBufferThreadFunc):wait timeout
06-23 19:46:24.391  3166 17570 W ExynosCamera3: [CAM_ID(0)][]-WARN(m_previewStream3AAPipeThreadFunc):wait timeout
06-23 19:46:24.501  3166 17566 W ExynosCameraMCPipe: [CAM_ID(0)][PIPE_3AA]-WARN(m_putBuffer[1392]):inputFrameQ wait timeout
06-23 19:46:24.501  3166 17566 W ExynosCameraMCPipe: [CAM_ID(0)][PIPE_3AA]-WARN(m_putBufferThreadFunc[1348]):m_putbuffer fail, ret(-110)
06-23 19:46:24.541  3166 17571 W ExynosCamera3: [CAM_ID(0)][]-WARN(m_previewStreamISPPipeThreadFunc):wait timeout
06-23 19:46:24.541  3166 17572 W ExynosCamera3: [CAM_ID(0)][]-WARN(m_duplicateBufferThreadFunc):wait timeout
06-23 19:46:24.541  3166 17570 W ExynosCamera3: [CAM_ID(0)][]-WARN(m_previewStream3AAPipeThreadFunc):wait timeout
06-23 19:46:24.551  3166 17568 W ExynosCameraMCPipe: [CAM_ID(0)][PIPE_ISP]-WARN(m_putBuffer[1392]):inputFrameQ wait timeout
06-23 19:46:24.551  3166 17568 W ExynosCameraMCPipe: [CAM_ID(0)][PIPE_ISP]-WARN(m_putBufferThreadFunc[1348]):m_putbuffer fail, ret(-110)
06-23 19:46:24.601  3166 17567 W ExynosCameraMCPipe: [CAM_ID(0)][PIPE_3AA]-WARN(m_getBuffer[1696]):requestFrameQ wait timeout
06-23 19:46:24.601  3166 17567 W ExynosCameraMCPipe: [CAM_ID(0)][PIPE_3AA]-WARN(m_getBufferThreadFunc[1364]):m_getBuffer fail, ret(-110)
06-23 19:46:24.601  3166 17569 W ExynosCameraMCPipe: [CAM_ID(0)][PIPE_ISP]-WARN(m_getBuffer[1696]):requestFrameQ wait timeout
06-23 19:46:24.601  3166 17569 W ExynosCameraMCPipe: [CAM_ID(0)][PIPE_ISP]-WARN(m_getBufferThreadFunc[1364]):m_getBuffer fail, ret(-110)
06-23 19:46:24.691  3166 17572 W ExynosCamera3: [CAM_ID(0)][]-WARN(m_duplicateBufferThreadFunc):wait timeout
06-23 19:46:24.691  3166 17571 W ExynosCamera3: [CAM_ID(0)][]-WARN(m_previewStreamISPPipeThreadFunc):wait timeout
06-23 19:46:24.691  3166 17570 W ExynosCamera3: [CAM_ID(0)][]-WARN(m_previewStream3AAPipeThreadFunc):wait timeout
06-23 19:46:24.801  3586  4695 I EDMNativeHelperService: isCameraEnabled
06-23 19:46:24.841  3166 17572 W ExynosCamera3: [CAM_ID(0)][]-WARN(m_duplicateBufferThreadFunc):wait timeout
06-23 19:46:24.841  3166 17571 W ExynosCamera3: [CAM_ID(0)][]-WARN(m_previewStreamISPPipeThreadFunc):wait timeout
06-23 19:46:24.841  3166 17570 W ExynosCamera3: [CAM_ID(0)][]-WARN(m_previewStream3AAPipeThreadFunc):wait timeout
[... cut ...]
06-23 19:46:25.151  3166 17567 W ExynosCameraMCPipe: [CAM_ID(0)][PIPE_3AA]-WARN(m_getBuffer[1696]):requestFrameQ wait timeout
06-23 19:46:25.151  3166 17567 W ExynosCameraMCPipe: [CAM_ID(0)][PIPE_3AA]-WARN(m_getBufferThreadFunc[1364]):m_getBuffer fail, ret(-110)
06-23 19:46:25.151  3166 17569 W ExynosCameraMCPipe: [CAM_ID(0)][PIPE_ISP]-WARN(m_getBuffer[1696]):requestFrameQ wait timeout
06-23 19:46:25.151  3166 17569 W ExynosCameraMCPipe: [CAM_ID(0)][PIPE_ISP]-WARN(m_getBufferThreadFunc[1364]):m_getBuffer fail, ret(-110)```

jawadsdsol, 2017-07-13:
Go the Camera2 class in library section and comment those lines in captureStillPicture() function.
/////////////////////////////////////////////////////////////////////////////////////////////////
 case Constants.FLASH_ON:
                    captureRequestBuilder.set(CaptureRequest.CONTROL_AE_MODE,
                            CaptureRequest.CONTROL_AE_MODE_ON_ALWAYS_FLASH);

                    break;
////////////////////////////////////////////////////////////////////////////////////////////////
That's work for me. try it.

skmll, 2017-10-26:
I was able to workaround this by NOT stopping/starting the preview capture request when taking a picture. I was looking into Camera2Raw example where I did not have this issue, and I noticed they dont stop the preview there, so I tried it and it worked. Basically comment this lines:
method unlockFocus():
// mCaptureSession.setRepeatingRequest(mPreviewRequest, mCaptureCallback,
// mBackgroundHandler);

method captureStillPicture():
// mCaptureSession.stopRepeating();
// mCaptureSession.abortCaptures();