google/cameraview #173: Crash on Sony Z5 Compact

ramdroid, 2017-06-23:
When switching few times between front/back camera the camera app freezes. When in this state the user then hits the flash point then the following exception is thrown:

```
STACK_TRACE=java.lang.NullPointerException: Attempt to invoke virtual method 'void android.hardware.camera2.CaptureRequest$Builder.set(android.hardware.camera2.CaptureRequest$Key, java.lang.Object)' on a null object reference
at com.google.android.cameraview.Camera2.lockFocus(Camera2.java:583)
at com.google.android.cameraview.Camera2.takePicture(Camera2.java:338)
at com.google.android.cameraview.CameraView.takePicture(CameraView.java:407)
at com.fastprotect.terra4d.tracker.CameraFragment.lambda$-com_fastprotect_terra4d_tracker_CameraFragment_lambda$0(CameraFragment.java:79)
at com.fastprotect.terra4d.tracker.-$Lambda$12$xdgV476pF-AFa1Kt759gG35Ld4E.$m$0(Unknown Source)
at com.fastprotect.terra4d.tracker.-$Lambda$12$xdgV476pF-AFa1Kt759gG35Ld4E.onClick(Unknown Source)
at android.view.View.performClick(View.java:5657)
at android.view.View$PerformClick.run(View.java:22314)
at android.os.Handler.handleCallback(Handler.java:751)
at android.os.Handler.dispatchMessage(Handler.java:95)
at android.os.Looper.loop(Looper.java:241)
at android.app.ActivityThread.main(ActivityThread.java:6217)
at java.lang.reflect.Method.invoke(Native Method)
at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:865)
at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:755)

```

dilix, 2017-06-30:
This happen because session is not initialized yet but you try to 'take picture'.
It's easy to reproduce on many devices by pressing 'rotate camera' + 'take picture' at the same time.

This fix helps me:
https://github.com/google/cameraview/pull/175

ramdroid, 2017-07-10:
Sorry, didn't work for me.

I tried with the patch but still crashes in the same function. Now also easily reproducible with the instructions you wrote on my Nexus 5 and it crashes at the same place too.

Also tested on a phone with Android 4.1.1 (camera1 interface) and it didn't crash but it lagged for a while and then refused to take any pictures after rotating camera... until I reopened the camera activity.

Gurupreet, 2017-08-03:
app is crashing on asus zenfone android 5.0.2  on changing aspect ratio and also showing black screen for some aspect ratio.
thanks