google/cameraview #183: Black CameraView after onResume only in Android 7

gmerwan, 2017-07-24:
Black CameraView after onResume (after onPause) only in Android 7. The camera is taking photo but the view is always black when returning to activity. (ONLY for Android Nougat 7)

ukevgen, 2017-07-25:
Yes, how we can restore camera in this case?

ukevgen, 2017-07-25:
Camera on full screen

gmerwan, 2017-08-01:
Any updates for that ?

Gurupreet, 2017-08-03:
app is crashing on asus zenfone android 5.0.2  on changing aspect ratio and also showing black screen for some aspect ratio.
thanks

jaksab, 2017-08-09:
I have this problem too

gmerwan, 2017-08-09:
We are waiting for an answer or fix ;)

serghov, 2017-08-27:
same problem

smellouk, 2017-09-25:
same issue here

smellouk, 2017-09-25:
same issue

smellouk, 2017-09-26:
Hi, I notice the black screen is triggered only when the library try to start Camera2. It fails because the device does not support hardware layer, so it will fallback to `Camera1` with `TextureViewPreview`, I think the issue is related to `TextureView`

**"TextureView can only be used in a hardware accelerated window. When rendered in software, TextureView will draw nothing"**

As fix or workaround for that  I adjust `createPreviewImpl` to return `SurfaceViewPreview` when the `CameraView` try to fallback to `Camera1` 

`@NonNull`
`private PreviewImpl createPreviewImpl(Context context, boolean isFallback) {`
`        if(isFallback)`
`            return new SurfaceViewPreview(context, this);`
`        PreviewImpl preview;`
`        if (Build.VERSION.SDK_INT < 14) {`
`            preview = new SurfaceViewPreview(context, this);`
`        } else {`
`            preview = new TextureViewPreview(context, this);`
`        }`
`        return preview;`
`}`

yangdingguo, 2017-11-30:
I had the same problrem,   VivO x9  7.1.1 I found that  the PowerKey is OK ,but the homeKey do not work well,any one have the resolution?

yangdingguo, 2017-11-30:
@smellouk   thank you for your advice    but I  7.1.1 use TextureView  ,can you say some detail

gmerwan, 2017-11-30:
@smellouk Can you provide us with the solution ... I mean the modified Library. 
Thank you.

yangdingguo, 2017-11-30:
@gmerwan 
public void start() {
        if ( !mImpl.start() ) {
            //store the state ,and restore this state after fall back o Camera1
            Parcelable state = onSaveInstanceState();
            // Camera2 uses legacy hardware layer; fall back to Camera1  then must use Surfaceview ;or black screen
            mImpl = new Camera1(mCallbacks, new SurfaceViewPreview(getContext(), this));

            onRestoreInstanceState(state);
            mImpl.start();
        }
    }

replace the method  in CameraView.Java with this method,maybe  it works well

gmerwan, 2017-11-30:
@yangdingguo gonna try it ... Thanks ;)

smellouk, 2017-12-01:
@gmerwan @yangdingguo 
Hey guys, 
I'm using oneplus with android nougat and I was facing the same issue but as I said a fix for that is [here](https://github.com/google/cameraview/issues/183#issuecomment-332142175)

So how you can Implement that ? 
You just need go to the [`CameraView.class`]  and in the line [246](https://github.com/google/cameraview/blob/master/library/src/main/java/com/google/android/cameraview/CameraView.java#L246) and replace this [`createPreviewImpl(getContext())`] by this  [`createPreviewImpl(getContext(), true)`]

gmerwan, 2017-12-01:
@smellouk  Thank you very much
I already replace it ... gonna test it asap

PonnamKarthik, 2018-01-25:
Thanks @smellouk  for the solutions its working fine

champrocks3190, 2018-04-02:
but cameraview class is not editable or writable . it states when try to make change .

champrocks3190, 2018-04-02:
no option left may be except recreate activity .... 

boolean created=false;

 @Override
    protected void onResume() {
        super.onResume();
        Handler handler = new Handler();
        handler.postDelayed(new Runnable() {
            @Override
            public void run() {

                if (created) {
                    recreate();
                }
                cameraView.start();
                created = false;
            }
        }, 100);
    }

    @Override
    protected void onPause() {
        cameraView.stop();
        created = true;
        super.onPause();
    }

yangdingguo, 2018-04-02:
@smellouk   can you give  somemore code ?  do  you  have  the method  createPreviewImpl(context,boolean) with two parameters?

gmerwan, 2018-04-02:
@yangdingguo @champrocks3190 
 @smellouk 's solution works well
I tried it and it works.

yangdingguo, 2018-04-02:
@gmerwan  @champrocks3190  @smellouk    can  you give  the method createPreviewImpl(context,boolean)  ?  now  I  have  no  this method with two parameters

yangdingguo, 2018-04-02:
@PonnamKarthik 
can you give the method createPreviewImpl(context,boolean) ? now I have no this method with two parameters  ,thank  you

gmerwan, 2018-04-02:
@yangdingguo use this repo ... I changed the method and you can use it directly.
You can see the changes .. it works.

yangdingguo, 2018-04-02:
where can  I  see the changes? the source code have changed?

yangdingguo, 2018-04-02:
@gmerwan   where can I see the changes code  ? the source code have changed?

gmerwan, 2018-04-02:
sorry this repo https://github.com/gmerwan/cameraview

gmerwan, 2018-04-02:
@yangdingguo I changed the CameraView.class like @smellouk proposed.
Try it and tell me if there is any problem.

GandhiRujul, 2018-04-05:
@gmerwan 
when i am stop camera and start it again i am still facing the black screen issue..

gmerwan, 2018-04-05:
@GandhiRujul I'll see about that. have you tried this repo ?
https://github.com/gmerwan/cameraview
Thank you.

GandhiRujul, 2018-04-06:
@gmerwan ,  
No I was using that Google Camera.. Its working on your repository thank you so much.

gmerwan, 2018-04-06:
You're welcome @GandhiRujul

GandhiRujul, 2018-04-06:
@gmerwan 
I have used your repository but its working almost all devices but I noticed one thing, when I am capturing image in Samsung , Your method `onPictureTaken(CameraView cameraView,byte[] data)` is giving me rotated `byte[]`.  So when I am creating image on based on `byte[]` , I am getting Rotated Image in Samsung device.

gmerwan, 2018-04-06:
@GandhiRujul for the created image you can rotate it before saving it as a file, I used that somewhere before but forgot it. I suggest u use the camerakit repo:  [https://github.com/CameraKit/camerakit-android](url)
I'm using it personally and it got everything u need + more features u may need too.

As I said there is a way to rotate a bitmap before saving it but that would take more time for you to capture rotate and then save, so try the repo I suggested and tell me you opinion. hope it helps

GandhiRujul, 2018-04-06:
@gmerwan 
Actually I am not getting bitmap as per aspect ratio in CameraKit , that's why I have decided to change the library to google camera. 

I have already code for rotate bitmap but If I am holding two devices front of face and try to taking capture image (by back camera) so in samsung device i am getting already rotated bitmap(-90 before my rotation of bitmap) where as in nexus device i got the perfect bitmap(0 before my rotation of bitmap).

So after rotation both images orientation will be different than each other.

gmerwan, 2018-04-06:
@GandhiRujul I understand. I'll see what I can do and respond asap.

gmerwan, 2018-04-06:
@GandhiRujul  using camerakit I could get either byte array or bitmap object.

GandhiRujul, 2018-04-06:
@gmerwan 
But I need bitmap into specific aspect ratio so is it possible to set aspect ratio in camera kit ?

yangdingguo, 2018-04-08:
@gmerwan   I have seen the code  but I did not  try. I believe it can work well.your method is same with my method,but  we deal it in different   position.

GandhiRujul, 2018-04-09:
@gmerwan Did you got the issue of samsung device ( byte[] is getting rotated )?

pritesh25, 2018-07-20:
thanks @smellouk for your efforts
i solved above issue with the help of below code

[tested in actual device] lenovo , honor, samsung, one plus 

pass **true** as a parameter in below method and use that below value as 

    ```
@NonNull
    private PreviewImpl createPreviewImpl(Context context, boolean isFallback) {

        if(isFallback)
            return new SurfaceViewPreview(context, this);

        PreviewImpl preview;
        if (Build.VERSION.SDK_INT < 14) {
            preview = new SurfaceViewPreview(context, this);
        } else {
            preview = new TextureViewPreview(context, this);
        }
        return preview;
    }
```

gmerwan, 2019-01-07:
@ankuagarwal  Yeah maybe it doesn't use the camera2, but it works perfectly. they moved to version 1.0 so it should be better after the stable version. It is still in dev

changjiajie, 2019-01-24:
@yangdingguo mImpl = new Camera1(mCallbacks, new SurfaceViewPreview(getContext(), this));
This method returns java.lang.NullPointerException, It is mSurfaceView.getHolder().
Try this solution：
@Override
protected void onResume(){
        if (mCameraRootV != null && mCameraV != null) {
            mCameraRootV.removeView(mCameraV);
            mCameraRootV.addView(mCameraV);
        }
        mCameraV.onResume(mPresent);
}

thaodersar, 2019-03-02:
Hi I try solution @smellouk shared above and it works. As you said it is workaround solution since TextureView and SurfaceView is just a view for streaming camera preview, and not related when camera 2 failed to start. 

I suppose that the current TextureView still keep streaming with camera hardware when you resume and this time you create a new TextureView with createPreviewImpl. Maybe there is an issue on android 7 that cannot replace the TextureView. I try this solution and it works, resume speed is very quick:

Create an variable to keep textureview instance: PreviewImpl mPreviewImpl;
then modify start like this:

public void start() {
        if (!mImpl.start()) {
            Log.d("Camera2", "start camera 2 fail");
            //store the state ,and restore this state after fall back o Camera1
            Parcelable state=onSaveInstanceState();
            // Camera2 uses legacy hardware layer; fall back to Camera1
//            mImpl = new Camera1(mCallbacks, createPreviewImpl(getContext(), true));
            mImpl = new Camera1(mCallbacks, mPreviewImpl);
            onRestoreInstanceState(state);
            mImpl.start();
        }
    }

Hope this help, I cannot test on other devices & android version so you guys can help

chinalwb, 2019-09-24:
> Hi I tried solution @smellouk shared above and it works. As you said it is workaround solution since TextureView or SurfaceView is just a view for streaming camera preview, and not related when camera 2 failed to start (I tested screen is black but camera still work).
> 
> I suppose that the current TextureView still keep streaming with camera hardware when you resume and this time you create a new TextureView with createPreviewImpl. Maybe there is an issue on android 7 that cannot replace the TextureView. I try this solution and it works perfectly, resume speed is very quick:
> 
> Create an variable to keep textureview instance: PreviewImpl mPreviewImpl;
> 
> Update CameraView constructor:
> .....
> // Internal setup
> mPreviewImpl = createPreviewImpl(context, false);
> mCallbacks = new CallbackBridge();
> if (Build.VERSION.SDK_INT < 21) {
> mImpl = new Camera1(mCallbacks, mPreviewImpl);
> } else if (Build.VERSION.SDK_INT < 23) {
> mImpl = new Camera2(mCallbacks, mPreviewImpl, context);
> } else {
> mImpl = new Camera2Api23(mCallbacks, mPreviewImpl, context);
> }
> ....
> 
> finally modify start() like this:
> 
> public void start() {
> if (!mImpl.start()) {
> Log.d("Camera2", "start camera 2 fail");
> //store the state ,and restore this state after fall back o Camera1
> Parcelable state=onSaveInstanceState();
> // Camera2 uses legacy hardware layer; fall back to Camera1
> // mImpl = new Camera1(mCallbacks, createPreviewImpl(getContext(), true));
> mImpl = new Camera1(mCallbacks, mPreviewImpl);
> onRestoreInstanceState(state);
> mImpl.start();
> }
> }
> 
> Hope this help, I cannot test on other devices & android versions so you guys can help

You saved me, sir! Thanks very much!

closed 2019-09-26 by gmerwan