google/cameraview #184: CameraView freeze when using the Front camera (Device: Sony Xperia XZ (F8332))

magiclen, 2017-07-28:
07-28 17:32:15.780: E/camera-hal(2518): camera_device.cpp(3750):validateCaptureSettings: Unsupported ae mode 2.
	07-28 17:32:15.780: E/camera-hal(2518): camera_device.cpp(5565):validateCaptureSettings: validateCaptureSettings failed.
	07-28 17:32:15.780: E/camera-hal(2518): camera_device.cpp(2869):processRequest: Invalid settings for capture. Device:1 Frame Number:0
	07-28 17:32:15.442: I/CameraService(2518): disconnect: Disconnected client for camera 0 for PID 25183
	07-28 17:32:15.447: I/CameraService(2518): CameraService::connect call (PID -1 "city.alley.alleycity", camera ID 1) for HAL version default and Camera API version 2
	07-28 17:32:15.448: I/Camera2ClientBase(2518): Camera 1: Opened. Client: city.alley.alleycity (PID 25183, UID 10250)
	07-28 17:32:15.448: I/CameraDeviceClient(2518): CameraDeviceClient 1: Opened
	07-28 17:32:15.450: D/audio_hw_primary(2466): adev_set_parameters: enter: camera_sub=on
	07-28 17:32:15.450: D/audio_hw_extn(2466): audio_extn_set_anc_parameters: anc_enabled:0
	07-28 17:32:15.451: W/ServiceManager(2518): Permission failure: com.sonyericsson.permission.CAMERA_EXTENDED from uid=10250 pid=25183
	07-28 17:32:15.451: W/ServiceManager(2518): Permission failure: com.sonymobile.permission.CAMERA_DUAL_OPEN from uid=10250 pid=25183
	07-28 17:32:15.452: W/ServiceManager(2518): Permission failure: com.sonymobile.permission.CAMERA_LOW_POWER from uid=10250 pid=25183
	07-28 17:32:15.452: I/camera-hal(2518): camera_module.cpp(120):registerCamera: open was requested by non Sony camera app

I've tried to assign front facing by using both `setFacing` method and `app:facing` XML attribute, the CameraView freezes all the time and cannot take a picture.

PS. The back camera works well. Camera API v1 + SurfaceView seems to be quite normal with the front camera.

Gurupreet, 2017-08-03:
app is crashing on asus zenfone android 5.0.2  on changing aspect ratio and also showing black screen for some aspect ratio.
thanks

kerimovscreations, 2017-08-24:
With my Xperia XZ, I'm encountering the same problem :(

crontab, 2017-08-24:
CameraView has a problem with pretty much all Sony Xperia compact series phones: closing and re-opening the camera causes the API to return an error (4) and consequently crashing because the internal camera object becomes null. This happens e.g. when you switch the cameras from back to front, change aspect ration etc. The OP's problem here may or may not be related though.

My workaround was to force Camera1 API on these phones. It may be that camera hardware on these phones are not fully Camera2 compliant despite that the phone itself officially supports API 21+.

kerimovscreations, 2017-08-24:
Agree. I have tested on Xperia Z Api level 21, it works great with Camera1. But Xperia XZ encounters with such kind of problems

TheMelody, 2017-09-19:
They code called when start the camera inside the **chooseCameraIdByFacing()** this method, no matter how you set the front-facing camera no, unless you call alone cameramanager. OpenCamera, (note) : **chooseCameraIdByFacing()** this method was sick, you can put it under renovation, I am after modification, and to use, switch the front will be a problem.I actually did it himself set up front rear logo, and then dynamically setting in it.

siralam, 2017-12-13:
I have tried to implement Camera2 API by myself, and worked on my Sony Xperia X Compact. I think falling back to Camera1 is not a good idea, actually the facing can be changed easily. I think as @TheMelody suggested, rewriting chooseCameraIdByFacing is a better approach.

siralam, 2017-12-14:
Today I tried in my real project, the freeze problem reappears....
This is really strange. I suppose we should go with @crontab 's suggestion of forcing Camera1 on Xperia series.

You may find this library useful: 
https://github.com/jaredrummler/AndroidDeviceNames

ruhy99, 2017-12-14:
@siralam You suggest to me to use CameraView or CameraViewPlus. I still have the problem in my Xperia X Compact. I saw this thread yesterday so I tried to implement the CameraViewPlus but I didn't see changes in result. I tried other libs like CameraKit but I had a lot of problems too. Thanks for your contribution.

ruhy99, 2017-12-14:
@siralam ¿How can I use https://github.com/jaredrummler/AndroidDeviceNames to force Camera1?

siralam, 2017-12-14:
@ruhy99 CameraViewPlus is my own forked version, not released yet. But as I have said too, I tried it in my real project and faced the same problem. So I think it is really the best to force it change to Camera1.

> How can I use https://github.com/jaredrummler/AndroidDeviceNames to force Camera1?

You can't, with Google's current CameraView. They do not provide such option.  
You may fork and implement it by yourself, or wait for me to publish my first version of CameraViewPlus...  
The library I have suggest is only to help you find out the name of the device. (So, if it is an Xperia Compact, then start your logic of forcing to use Camera1)

siralam, 2017-12-14:
@ruhy99 By the way, if you want a good library without the need to use Camera2 API, I would suggest [this one](https://github.com/natario1/CameraView/issues/116)

ruhy99, 2017-12-14:
@siralam Thanks for the answers I still do not have the knowledge to use this AndroidDeviceName library to do this as well. I will wait an update to your fork if I don't have any camera library that fits mine project.

ruhy99, 2017-12-14:
@siralam Wow natario1/CameraView is so good it fits with my expectatives. It's easy to implement.

siralam, 2017-12-14:
@ruhy99 Yes, it is the best I can find.

But I abandoned it in my project due to [this issue](https://github.com/natario1/CameraView/issues/116), but not sure if you can reproduce it.
And since that library does not use Camera2, you will hear a shutter sound if the device has API level >= 21. Its OK if it does not bother you though.

Anyway, let's avoid talking too much about another library here. Feel free to email me if you want to continue the conversation.

kerimovscreations, 2017-12-14:
Guys, I'm using 'io.fotoapparat.fotoapparat' library. It works better than other camera libraries :)

ruhy99, 2017-12-14:
kerimovscreations I tried io.fotoapparat.fotoapparat in the past but I prefer the natario1/CameraView because I think that camera1 is robuster.

kerimovscreations, 2017-12-14:
@ruhy99 natario1/CameraView is not better because I tried this and other libraries on tablet, only 'io.fotoapparat.fotoapparat' has worked correctly (orientation problem was on tablet Xperia Z tablet) 😄

siralam, 2017-12-14:
@kerimovscreations May I ask where do you find fotoapparat? If I can find it earlier, it probably would have saved me huge amount of time :(

kerimovscreations, 2017-12-15:
@siralam I don't remember exactly, I searched camera libraries and tested all of them on my devices. Then decided to choose this library. I hope this will help other developers for further projects 😄

kraa, 2018-04-20:
What helps me is 
`app:flash="off"`
(problem with that in Front facing)
Thats all! After that facing is working in xml and in code. Error 1:(4) disappear

yanggq, 2018-08-10:
Try to remove  code from  your cam lib like the  following  line:

 `mCaptureRequestBuilder.set(CaptureRequest.COLOR_CORRECTION_ABERRATION_MODE, CameraMetadata.COLOR_CORRECTION_ABERRATION_MODE_HIGH_QUALITY);`

This works for me :-)