google/cameraview #185: 小米5plus 不能切换前置摄像头

tqdream, 2017-08-02:
小米5plus 不能切换前置摄像头

daquexian, 2017-08-02:
I think you should use English here :\ No one is responsible for spending time learning Chinese. And I think some details(like the version and the ROM of your phone) are needed.

-------------
Translation:
Front camera is not working on Xiaomi 5 Plus.

tqdream, 2017-08-07:
你好，我发现您项目中有一个方法写的有问题，Camera1.java文件152行代码
    @Override
    Set<AspectRatio> getSupportedAspectRatios() {
        SizeMap idealAspectRatios = mPreviewSizes;
        for (AspectRatio aspectRatio : idealAspectRatios.ratios()) {
            if (mPictureSizes.sizes(aspectRatio) == null) {
                idealAspectRatios.remove(aspectRatio);
            }
        }
        return idealAspectRatios.ratios();
    }
从集合中删除元素时有问题。

移除元素应该这样:
    @Override
    Set<AspectRatio> getSupportedAspectRatios() {
        SizeMap idealAspectRatios = mPreviewSizes;
        Iterator<AspectRatio> iterator = idealAspectRatios.ratios().iterator();
        while (iterator.hasNext()) {
            AspectRatio aspectRatio = iterator.next();
            if (mPictureSizes.sizes(aspectRatio) == null) {
                iterator.remove();
            }
        }
        return idealAspectRatios.ratios();
    }

TheMelody, 2017-09-04:
小米5S已经是6.0之上的手机了,Build.Version.SDK_INT>=21的话，需要通过CameraManager的openCamera方法设置前置摄像头和后置摄像头,同时cameraId为String类型,前置为"1",后置为"0".
mCameraManager.openCamera(cameraId,mCameraDeviceCallback,null);
加我的QQ群交流技术: 490113234

jimmysuncpt, 2018-12-15:
> 你好，我发现您项目中有一个方法写的有问题，Camera1.java文件152行代码
>
> ```
>     @Override
>     Set<AspectRatio> getSupportedAspectRatios() {
>         SizeMap idealAspectRatios = mPreviewSizes;
>         for (AspectRatio aspectRatio [...]

Camera2代码中collectCameraInfo()方法里也有这样的问题：
```
        for (AspectRatio ratio : mPreviewSizes.ratios()) {
            if (!mPictureSizes.ratios().contains(ratio)) {
                mPreviewSizes.remove(ratio);
            }
        }
```
应该改为：
```
        Iterator<AspectRatio> iterator = mPreviewSizes.ratios().iterator();
        while (iterator.hasNext()) {
            AspectRatio ratio = iterator.next();
            if (!mPictureSizes.ratios().contains(ratio)) {
                iterator.remove();
            }
        }
```

jimmysuncpt, 2018-12-16:
有个临时的解决方案：
在Camera2.java的chooseCameraIdByFacing()方法里368行前后：
```
                if (level == null ||
                        level == CameraCharacteristics.INFO_SUPPORTED_HARDWARE_LEVEL_LEGACY) {
                    continue;
                }
```
改成以下代码即可解决小米5s不能切换前置摄像头问题：
```
                if (level == null || 
                        level == CameraCharacteristics.INFO_SUPPORTED_HARDWARE_LEVEL_LEGACY) {
                    if (!Build.MODEL.equals("MI 5s")) {
                        continue;
                    }
                }
```
如果发现其它机型也出现类似问题，可以再增加条件判断。