google/cameraview #193: 华为P6 Android4.4闪退

hxk1224, 2017-08-17:
代码：方法onOptionsItemSelected()
case R.id.switch_camera:
                if (mCameraView != null) {
                    int facing = mCameraView.getFacing();
                    mCameraView.setFacing(facing == CameraView.FACING_FRONT ?
                            CameraView.FACING_BACK : CameraView.FACING_FRONT);//here
                }

08-17 11:37:09.400 6219-6219/com.google.android.cameraview.demo E/AndroidRuntime: FATAL EXCEPTION: main
                                                                                  Process: com.google.android.cameraview.demo, PID: 6219
                                                                                  java.lang.RuntimeException: setParameters failed
                                                                                      at android.hardware.Camera.native_setParameters(Native Method)
                                                                                      at android.hardware.Camera.setParameters(Camera.java:1671)
                                                                                      at com.google.android.cameraview.Camera1.adjustCameraParameters(Camera1.java:342)
                                                                                      at com.google.android.cameraview.Camera1.openCamera(Camera1.java:307)
                                                                                      at com.google.android.cameraview.Camera1.start(Camera1.java:90)
                                                                                      at com.google.android.cameraview.Camera1.setFacing(Camera1.java:142)
                                                                                      at com.google.android.cameraview.CameraView.setFacing(CameraView.java:319)
                                                                                      at com.google.android.cameraview.demo.MainActivity.onOptionsItemSelected(MainActivity.java:212)
                                                                                      at android.app.Activity.onMenuItemSelected(Activity.java:2614)
                                                                                      at android.support.v4.app.FragmentActivity.onMenuItemSelected(FragmentActivity.java:408)
                                                                                      at android.support.v7.app.AppCompatActivity.onMenuItemSelected(AppCompatActivity.java:195)
                                                                                      at android.support.v7.view.WindowCallbackWrapper.onMenuItemSelected(WindowCallbackWrapper.java:113)
                                                                                      at android.support.v7.view.WindowCallbackWrapper.onMenuItemSelected(WindowCallbackWrapper.java:113)
                                                                                      at android.support.v7.app.ToolbarActionBar$2.onMenuItemClick(ToolbarActionBar.java:69)
                                                                                      at android.support.v7.widget.Toolbar$1.onMenuItemClick(Toolbar.java:206)
                                                                                      at android.support.v7.widget.ActionMenuView$MenuBuilderCallback.onMenuItemSelected(ActionMenuView.java:776)
                                                                                      at android.support.v7.view.menu.MenuBuilder.dispatchMenuItemSelected(MenuBuilder.java:822)
                                                                                      at android.support.v7.view.menu.MenuItemImpl.invoke(MenuItemImpl.java:156)
                                                                                      at android.support.v7.view.menu.MenuBuilder.performItemAction(MenuBuilder.java:969)
                                                                                      at android.support.v7.view.menu.MenuBuilder.performItemAction(MenuBuilder.java:959)
                                                                                      at android.support.v7.widget.ActionMenuView.invokeItem(ActionMenuView.java:623)
                                                                                      at android.support.v7.view.menu.ActionMenuItemView.onClick(ActionMenuItemView.java:154)
                                                                                      at android.view.View.performClick(View.java:4444)
                                                                                      at android.view.View$PerformClick.run(View.java:18440)
                                                                                      at android.os.Handler.handleCallback(Handler.java:733)
                                                                                      at android.os.Handler.dispatchMessage(Handler.java:95)
                                                                                      at android.os.Looper.loop(Looper.java:136)
                                                                                      at android.app.ActivityThread.main(ActivityThread.java:5016)
                                                                                      at java.lang.reflect.Method.invokeNative(Native Method)
                                                                                      at java.lang.reflect.Method.invoke(Method.java:515)
                                                                                      at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:795)
                                                                                      at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:611)
                                                                                      at dalvik.system.NativeStart.main(Native Method)
08-17 11:37:09.415 6219-6219/com.google.android.cameraview.demo I/Process: Sending signal. PID: 6219 SIG: 9

TheMelody, 2017-09-19:
加我qq群：490113234 ，细聊，他们代码里面在启动相机的时候调用了chooseCameraIdByFacing这个方法，不论你怎么设置前置摄像头都不行，除非你单独调用cameramanager.openCamera，（注意）：chooseCameraIdByFacing这个方法简直恶心，你可以把里面改造下，我就是改造完之后，才能用，切换前置没什么问题了。我其实自己做了设置前置后置的标志，然后在里面动态设置。

xiaanming, 2017-10-10:
/**
     * @return {@code true} if {@link #mCameraParameters} was modified.
     */
    private boolean setFlashInternal(int flash) {
        if (isCameraOpened()) {
            List<String> modes = mCameraParameters.getSupportedFlashModes();
            //flash mode setting is not supported
            if (modes == null) {
                mFlash = flash;
                return false;
            }

            String mode = FLASH_MODES.get(flash);
            if (modes != null && modes.contains(mode)) {
                mCameraParameters.setFlashMode(mode);
                mFlash = flash;
                return true;
            }
            String currentMode = FLASH_MODES.get(mFlash);
            if (!modes.contains(currentMode)) {
                mCameraParameters.setFlashMode(Camera.Parameters.FLASH_MODE_OFF);
                mFlash = Constants.FLASH_OFF;
                return true;
            }
            return false;
        } else {
            mFlash = flash;
            return false;
        }
    }

motify this function can fix it