google/cameraview #208: Image is rotated for some devices

sitannsu, 2017-10-03:
How to detect if image rotated for some devices.
For some devices like samsung image is auto rotatted  after photo taken.

petergayduchik, 2017-11-14:
bug with 5 years history.  Main problem its  - samsung saves photo always in landscape mode . 
Say i make photo in portrate with size: w=1000 h = 1900. But saved photo will be  with w=1900 h = 1000. If go to landscape than everything fine. Only bug with portrate mode. Im trying to fix this second day.

sranjith096, 2017-11-20:
@petergayduchik  Did you find a hack for it?

petergayduchik, 2017-11-21:
@sranjith096 [https://stackoverflow.com/questions/47261434/photo-rotated-from-camera-samsung-device/47261631?noredirect=1#comment81474720_47261631](url) here a my solution . Not sure its will work on all devices.

zweinz, 2018-01-26:
I'm also finding this to be the case on a new Pixel 2. Any updates on this issue?

MoustafaElsaghier, 2018-04-25:
I've fixed this issue by checking the width and height of the image
After long of debugging, I've found that the image is rotated by 90 degrees on devices like Samsung and sony so I've implemented this simple method for solving that issue, hope it solves yours 
```
private static int fixOrientation(Bitmap bitmap, boolean isPortraite) {
        if (isPortraite) { 
            if (bitmap.getWidth() > bitmap.getHeight()) {
                return 90;
            }
        } else if (bitmap.getWidth() < bitmap.getHeight()) {
            return 90;
        }
        return 0;
    }
```
if any one need clearfication for this issue I 'm here

tomasmaks, 2018-05-29:
MoustafaElsaghier your solution worked for me, gratz!

MoustafaElsaghier, 2018-05-29:
@tomasmaks  it's pleasure for me that it works with you.

Vico231088, 2018-09-07:
como puedo llamar a este metodo para que se ejecute 👍 
private static int fixOrientation(Bitmap bitmap, boolean isPortraite) {
        if (isPortraite) { 
            if (bitmap.getWidth() > bitmap.getHeight()) {
                return 90;
            }
        } else if (bitmap.getWidth() < bitmap.getHeight()) {
            return 90;
        }
        return 0;
    }