google/cameraview #223: The Exception ：Failed to start camera session

yanxinmiao, 2017-12-28:
I got this issue on HuaWei Phone , but only the first time to use camera .

And I  look up all of the issue, don't get solution .

```
 java.lang.RuntimeException: Failed to start camera session
                      at com.google.android.cameraview.Camera2.startCaptureSession(Camera2.java:511)
                      at com.google.android.cameraview.Camera2$1.onOpened(Camera2.java:76)
                      at android.hardware.camera2.impl.CameraDeviceImpl$1.run(CameraDeviceImpl.java:134)
                      at android.os.Handler.handleCallback(Handler.java:761)
                      at android.os.Handler.dispatchMessage(Handler.java:98)
                      at android.os.Looper.loop(Looper.java:156)
                      at android.app.ActivityThread.main(ActivityThread.java:6523)
                      at java.lang.reflect.Method.invoke(Native Method)
                      at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:942)
                      at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:832)
```

yanxinmiao, 2018-01-02:
Please closed this issue ， I has sloved it

Reference https://github.com/google/cameraview/issues/53

shellavitancol, 2018-08-30:
How did you solved it? I am experiencing the same error.

yyyAndroid, 2020-03-23:
I am experiencing the same error.